zephyrproject-rtos/zephyr #59608: Bluetooth: Controller: Not able to connect to device while PA synced to other device

Thalley, 2023-06-22:
**Describe the bug**
If we are periodically synced to device A, then we seem to have issues connecting to another device B. 

**To Reproduce**
Steps to reproduce the behavior:
1. Periodically sync to device A
2. Scan for and to connect to device B
3. See error

**Expected behavior**
It should be possible to connect to a device and maintain a PA sync. 

**Impact**
Showstopper for the LE Audio Broadcast Assistant role

**Logs and console output**
```
	[2023-06-14 01:35:25] PA synced to broadcaster with ID 0x00CD55 as sink 0x20044c80
	[2023-06-14 01:35:25] Sink 0x20044c80 is set as default
	[2023-06-14 01:35:25] PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random) synced, Interval 0x0100 (320000 us), PHY LE 2M, SD 0x0000, PAST peer
	[2023-06-14 01:35:25] Received BASE from sink 0x20044c80:
	[2023-06-14 01:35:25] Subgroup[0]:
	[2023-06-14 01:35:25] codec 0x06 cid 0x0000 vid 0x0000
	[2023-06-14 01:35:25] data_count 4
	[2023-06-14 01:35:25] data #0: type 0x01 len 1
	[2023-06-14 01:35:25] data #1: type 0x02 len 1
	[2023-06-14 01:35:25] data #2: type 0x03 len 4
	[2023-06-14 01:35:25] 00000000: 01 00 00                                         |...              |
	[2023-06-14 01:35:25] data #3: type 0x04 len 2
	[2023-06-14 01:35:25] 00000000: 3c                                               |<                |
	[2023-06-14 01:35:25] meta_count 4
	[2023-06-14 01:35:25] meta #0: type 0x02 len 2
	[2023-06-14 01:35:25] 00000000: 04                                               |.                |
	[2023-06-14 01:35:26] BIS[0] index 0x01
	[2023-06-14 01:35:26] data #0: type 0x01 len 1
	[2023-06-14 01:35:26] BIS[0] index 0x02
	[2023-06-14 01:35:26] data #0: type 0x01 len 1
	[2023-06-14 01:35:26] [0]: 0x01
	[2023-06-14 01:35:26] [1]: 0x02
	[2023-06-14 01:35:26] Possible indexes: 0x01 0x02
	[2023-06-14 01:35:26] PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random), tx_power 127, RSSI -51, CTE 0, data length 46
	[2023-06-14 01:35:26] Sink 0x20044c80 is ready to sync without encryption
	[2023-06-14 01:35:26] BIG_INFO PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random), sid 0x05, num_bis 2, nse 0x03, interval 0x0008 (10000 us), bn 0x01, pto 0x00, irc 0x03, max_pdu 0x003c, sdu_interval 0x2710, max_sdu 0x003c, phy LE 2M,
	framing 0x00, not encrypted
	[2023-06-14 01:35:26] [00:01:04.919,219] <dbg> bt_bap_broadcast_sink: broadcast_scan_recv: Found broadcast source with address 34:CD:C8:C0:0B:85 (random) and id 0x00CD55
	[2023-06-14 01:35:26] [00:01:05.401,641] <dbg> bt_bap_broadcast_sink: pa_synced: Synced to broadcast source with ID 0x00CD55
	[2023-06-14 01:35:26] [00:01:05.401,672] <dbg> bt_bap_broadcast_sink: bt_bap_broadcast_sink_scan_stop: Not scanning
	[2023-06-14 01:35:26] [00:01:05.427,795] <wrn> bt_bap_broadcast_sink: update_recv_state_base: Failed to find receive state for sink 0x20044c80
	[2023-06-14 01:35:26] BIG_INFO PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random), sid 0x05, num_bis 2, nse 0x03, interval 0x0008 (10000 us), bn 0x01, pto 0x00, irc 0x03, max_pdu 0x003c, sdu_interval 0x2710, max_sdu 0x003c, phy LE 2M,
	framing 0x00, not encrypted

	[2023-06-14 01:38:27] uart:~$ bt connect-name ceva
	[2023-06-14 01:38:36] Bluetooth active scan enabled
	[2023-06-14 01:38:36] uart:~$
	[2023-06-14 01:38:37] [DEVICE]: AB:89:67:45:23:01 (public), AD evt type 5, RSSI -67 CEVA_HEADSET_BC C:1 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 1M, Interval: 0x0000 (0 us), SID: 0x1
	[2023-06-14 01:38:42] Scan successfully stopped
	[2023-06-14 01:38:42] Connection pending
	[2023-06-14 01:38:45] Failed to connect to AB:89:67:45:23:01 (public) (0x02)
	[2023-06-14 01:38:45] Failed to connect to AB:89:67:45:23:01 (public) (2)
	[2023-06-14 01:38:45] [00:04:25.082,611] <dbg> bt_has: connected: conn 0x20018ac8 err 2
	[2023-06-14 01:38:46] uart:~$
	[2023-06-14 01:38:47] uart:~$
	[2023-06-14 01:38:47] uart:~$ bt connect-name ceva
	[2023-06-14 01:38:49] Bluetooth active scan enabled
	[2023-06-14 01:38:49] [DEVICE]: AB:89:67:45:23:01 (public), AD evt type 5, RSSI -67 CEVA_HEADSET_BC C:1 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 1M, Interval: 0x0000 (0 us), SID: 0x1
	[2023-06-14 01:38:55] Scan successfully stopped
	[2023-06-14 01:38:55] Connection pending
	[2023-06-14 01:38:55] BIG_INFO PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random), sid 0x05, num_bis 2, nse 0x03, interval 0x0008 (10000 us), bn 0x01, pto 0x00, irc 0x03, max_pdu 0x003c, sdu_interval 0x2710, max_sdu 0x003c, phy LE 2M,
	framing 0x00, not encrypted
	[2023-06-14 01:38:55] BIG_INFO PER_ADV_SYNC[0]: [DEVICE]: 34:CD:C8:C0:0B:85 (random), sid 0x05, num_bis 2, nse 0x03, interval 0x0008 (10000 us), bn 0x01, pto 0x00, irc 0x03, max_pdu 0x003c, sdu_interval 0x2710, max_sdu 0x003c, phy LE 2M,
	framing 0x00, not encrypted
	[2023-06-14 01:38:55] Failed to connect to AB:89:67:45:23:01 (public) (0x02)
	[2023-06-14 01:38:58] Failed to connect to AB:89:67:45:23:01 (public) (2)
	[2023-06-14 01:38:58] [00:04:38.295,013] <dbg> bt_has: connected: conn 0x20018ac8 err 2
	[2023-06-14 01:38:59] uart:~$
	[2023-06-14 01:39:00] uart:~$
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/59107

Using the nRF5340 with the hci_rpmsg sample with `nrf5340_cpunet_iso-bt_ll_sw_split.conf` on the netcore and the audio shell on the appcore. 

**Additional context**
Issue found during UPF, so found while there is significant network traffic.

cvinayak, 2024-01-30:
@Thalley I used the tests/bluetooth/shell on nrf52833dk, and able to have both a PA sync and be able to establish a connection to another device. Is my testing correct?

[image]

Thalley, 2024-01-30:
> @Thalley I used the tests/bluetooth/shell on nrf52833dk, and able to have both a PA sync and be able to establish a connection to another device. Is my testing correct? [...]

Yup, looks like it works for that case. I think we can close this issue, and see if we'll see anything similar in the future, but at least this particular one seems OK

closed 2024-01-30 by Thalley