zephyrproject-rtos/zephyr #59624: CAN: STM32H7: can2 interface is not usable

pfayolle, 2023-06-22:
**Describe the bug**
The can2 interface as defined in the STM32H7 device tree is not able to send or receive CAN frames.

I use a custom board, based on the STM32H743VI, with two CAN interfaces and have issues to make them work simultaneously.
I wrote a test program that sets up threads and messages queues, in the same way for each CAN interface (can1 and can2) so that any CAN frame received on one interface is sent back on the same interface while incrementing the first data byte of the payload.

The frames are in the extended format, and the receiving is handled with a filter that accepts all data frames in extended format, and uses a message queue.

Whether I connect the two interfaces together or use a setup with two boards, I always get the same result: can1 is able to send and receive CAN frames while can2 is not.

However, can2 works if I disable the can1 interface and alter the can2 interface definition in the STM32H7 device tree by replacing these two lines:
```
reg = <0x4000a400 0x400>, <0x4000ac00 0x6a0>;
bosch,mram-cfg = <0x350 28 8 3 3 0 3 3>;
```
with these two lines:
```
reg = <0x4000a400 0x400>, <0x4000ac00 0x350>;
bosch,mram-cfg = <0x0 28 8 3 3 0 3 3>;
```

This makes me think there could be an issue in the CAN driver related to the message RAM layout.

**To Reproduce**
- Set the can2 interface to okay in the STM32H7 device tree
- Build an app that receives all extended data frames on can2 and sends them back
- Connect the can2 interface to another working CAN interface that sends extended data frames, while making sure the bus speeds match on the two interfaces
- Observe that no CAN frame is received on CAN2

**Expected behavior**
can2 is able to send and receive CAN frames, while simultaneously using the can1 interface.

**Impact**
can2 is not usable on my board.

erwango, 2023-06-23:
@henrikbrixandersen, can I assign this to you ?

FRASTM, 2023-06-23:
Could this be due to a wrong definition of the` CAN Message RAM  : 0x4000AC00 - 0x4000D3FF`  (0x2800 bytes = 2560 * 4 ) which is not partitioned for each fdcan controller but common to all the controllers. Each of them defining its own Rx/Tx buffer start address 
[image]

henrikbrixandersen, 2023-06-23:
> @henrikbrixandersen, can I assign this to you ?

I'll be happy to take a look, but I don't have the hardware to test it, so it will be purely by data sheet and code inspection.

henrikbrixandersen, 2023-06-23:
> Could this be due to a wrong definition of the` CAN Message RAM : 0x4000AC00 - 0x4000D3FF` (0x2800 bytes = 2560 * 4 ) which is not partitioned for each fdcan controller but common to all the controllers. Each of them defining its own Rx/Tx buffer start address

The "partitioning" should be handled by the offset cell in the `bosch,mram-cfg` DTS property, e.g. for `can2` it is `bosch,mram-cfg = <0x350 28 8 3 3 0 3 3>;`, meaning an offset of `0x350`. So all STM32H7 CAN controllers use the same Message RAM, but use different, non-overlapping start addresses. Is this what you meant?

FRASTM, 2023-06-23:
> RAM, but use different, non-overlapping start addresses. Is this what you meant?

Yes more or less,  at first glance 
if for can1  `reg = <0x4000a000 0x400>, <0x4000ac00 0x350>;`
then for can2 `reg = <0x4000a400 0x400>, <4000AF50 0x350>;`
etc

pfayolle, 2023-07-04:
I've tried adjusting the addresses and offsets for can2 without success:

If I use ```reg = <0x4000a400 0x400>, <0x4000af50 0x350>;``` and  ```bosch,mram-cfg = <0x0 28 8 3 3 0 3 3>;``` the interface still does not work.
If I use ```reg = <0x4000a400 0x400>, <0x4000af50 0x350>;``` and  ```bosch,mram-cfg = <0x350 28 8 3 3 0 3 3>;```, I get the following build error: 
```
zephyr/drivers/can/can_stm32h7.c:197:9: error: static assertion failed: "Insufficient Message RAM size to hold elements"
  197 |         BUILD_ASSERT(CAN_MCAN_DT_INST_MRAM_ELEMENTS_SIZE(n) <=              \
```

henrikbrixandersen, 2023-07-05:
> I've tried adjusting the addresses and offsets for can2 without success:
>
> If I use `reg = <0x4000a400 0x400>, <0x4000af50 0x350>;` and `bosch,mram-cfg = <0x0 28 8 3 3 0 3 3>;` the interface still [...]

Right. With the configuration above, you have reduced the Bosch M_CAN message RAM size to zero bytes, hence the number of elements specified will not fit.

henrikbrixandersen, 2023-07-05:
> > RAM, but use different, non-overlapping start addresses. Is this what you meant?
> 
> Yes more or less, at first glance if for can1 `reg = <0x4000a000 0x400>, <0x4000ac00 0x350>;` then for can2 `reg = <0x4000a400 0x400>, <4000af50 0x350>;` etc

The offset cell in `bosch,mram_cfg` should take of the offset for the individual instance's offset within the given memory area.

pfayolle, 2023-07-21:
Is there anything I should try in my device tree to make it work ?

henrikbrixandersen, 2023-07-21:
> Is there anything I should try in my device tree to make it work ?

Apart from general the advice to use a debugger to find out where the bug is, I'm out of ideas. 

Luckily, I will soon have access to STM32H7 hardware, courtesy of @erwango, so I should be able to reproduce the issue and debug it.

henrikbrixandersen, 2023-07-28:
@pfayolle Please try the fix in #60935.

closed 2023-07-31 by carlescufi

pfayolle, 2023-07-31:
@henrikbrixandersen I just tested on my board and it works perfectly now. Thank you !

PR #60935 (2023-07-31): drivers: can: stm32h7: fix message RAM address calculations
Calculate the Bosch M_CAN Message RAM addresses relative to the Message RAM Base Address (MRBA), not the offset. Add `CAN_MCAN_DT_MRBA()` and `CAN_MCAN_DT_INST_MRBA()` helper macros.
    
Fixes: #59624
PR #60976 (2023-08-02): [Backport v3.4-branch] drivers: can: stm32h7: fix message RAM address calculations
Backport b809d5ce1049f541def9408112bef4e540f2a7b6~2..b809d5ce1049f541def9408112bef4e540f2a7b6 from #60935.

Fixes: #59624