zephyrproject-rtos/zephyr #59636: Instruction Access Violation -- MPU Fault

nkaufmn, 2023-06-22:
I have an issue on the send request of LoRaWAN.

I work with an NRF52840 with external NOR Memory (MX25L3233F), the partition "settings" is on external memory, and a SX1276 is used as LoRa module.  I use the NRF Connect SDK 2.4.0 on Visual Studio Code.

Behavior: My device could join to thethingsnetwork but when it sends a message to ttn it crashes with an fatal error. 

This is the DEBUG output:

> *** Booting Zephyr OS build v3.3.99-ncs1 ***
> Attempting to boot slot 0.
> Attempting to boot from address 0x9200.
> Verifying signature against key 0.
> Hash: 0x5b...a3
> Firmware signature verified.
> Firmware version 1
> Setting monotonic counter (version: 1, slot: 0)
> [00:00:00.000,152] <inf> board_control: Board configured.
> [00:00:00.200,866] <inf> sx127x: SX127x version 0x12 found
> --- 18 messages dropped ---
> Starting LoRaWAN stack.
> [00:00:00.360,260] <dbg> lorawan: lorawan_start: LoRaMAC Initialized
> [00:00:00.388,183] <inf> fs_nvs: 4 Sectors of 4096 bytes
> [00:00:00.388,183] <inf> fs_nvs: alloc wra: 2, f68
> [00:00:00.388,214] <inf> fs_nvs: data wra: 2, d08
> [00:00:00.393,615] <dbg> lorawan_nvm: lorawan_nvm_data_restore: Restoring LoRaWAN settings
> [00:00:00.404,693] <dbg> lorawan_nvm: on_setting_loaded: Key: ClassB
> [00:00:00.417,846] <dbg> lorawan_nvm: on_setting_loaded: Key: RegionGroup2
> [00:00:00.426,300] <dbg> lorawan_nvm: on_setting_loaded: Key: SecureElement
> [00:00:00.433,441] <dbg> lorawan_nvm: on_setting_loaded: Key: MacGroup2
> [00:00:00.440,460] <dbg> lorawan_nvm: on_setting_loaded: Key: MacGroup1
> [00:00:00.447,479] <dbg> lorawan_nvm: on_setting_loaded: Key: Crypto
> [00:00:00.448,059] <dbg> lorawan_nvm: lorawan_nvm_data_restore: Crypto version: 16777472, DevNonce: 112, JoinNonce: 333
> [00:00:00.449,188] <dbg> lorawan_nvm: lorawan_nvm_data_restore: LoRaWAN context restored
> Joining network using OTAA, dev nonce 0, attempt 1:
> [00:00:00.454,040] <dbg> lorawan: lorawan_join: Network join request sent!
> [00:00:05.591,552] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
> [00:00:05.591,583] <inf> lorawan: Joined network! DevAddr: 260b0264
> [00:00:05.592,834] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
> [00:00:05.592,864] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 113, JoinNonce: 334
> [00:00:05.592,895] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/Crypto
> New Datarate: DR_5, Max Payload 242[00:00:05.658,691] <inf> lorawan: Datarate changed: DR_5
> [00:00:05.628,997] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup1
> Join successful.
> [00:00:05.659,393] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup2
> LoRa init complete
> [00:00:05.685,791] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/SecureElement
> [00:00:10.958,068] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 0)
> [00:00:10.958,099] <dbg> lorawan: mcps_confirm_handler: McpsRequest success!
> [00:00:10.958,129] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
> [00:00:10.958,160] <err> os: ***** MPU FAULT *****
> [00:00:10.958,160] <err> os:   Instruction Access Violation
> [00:00:10.958,190] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0xffffffd6
> [00:00:10.958,221] <err> os: r3/a4:  0x00000008 r12/ip:  0xffffffff r14/lr:  0x0004959d
> [00:00:10.958,221] <err> os:  xpsr:  0x01000000
> [00:00:10.958,251] <err> os: Faulting instruction address (r15/pc): 0x20006d30
> [00:00:10.958,282] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
> [00:00:10.958,312] <err> os: Current thread: 0x20004448 (unknown)

I figured out, that the system crashes when it tries to do this:
The code snippet is from "lorawan.c" line 136 - 165

```
static void mcps_indication_handler(McpsIndication_t *mcps_indication)
{
	struct lorawan_downlink_cb *cb;

	LOG_DBG("Received McpsIndication %d", mcps_indication->McpsIndication);

	if (mcps_indication->Status != LORAMAC_EVENT_INFO_STATUS_OK) {
		LOG_ERR("McpsIndication failed : %s",
[... 176 lines cut ...]
> CONFIG_CBPRINTF_FP_SUPPORT=y
> CONFIG_SETTINGS=y
> CONFIG_NEWLIB_LIBC=y
> CONFIG_CPLUSPLUS=y

And the sx1276 is configured in the overlay file as followed:

> &spi1 {
> 	compatible = "nordic,nrf-spi";
> 	status = "okay";
> 	cs-gpios = <&gpio0 12 GPIO_ACTIVE_LOW>;
> 	pinctrl-0 = <&spi1_default>;
> 	pinctrl-1 = <&spi1_sleep>;
> 	pinctrl-names = "default", "sleep";
> 	lora: sx1276@0 {
>         compatible = "semtech,sx1276";
>         reg = <0>;
> 		reset-gpios = <&gpio0 14 GPIO_ACTIVE_LOW>; //reset-gpios = <&gpio1 8 GPIO_ACTIVE_LOW>;
>         dio-gpios = <&gpio0 11 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
>                     <&gpio0 21 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
>                     <&gpio0 25 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
> 	power-amplifier-output = "pa-boost";
> 	spi-max-frequency = <1000000>;
>     };
> };

Mani-Sadhasivam, 2023-06-23:
@nkaufmn Can you share the downlink callback that has been registered using `lorawan_register_downlink_callback()` in the sample application?

Mani-Sadhasivam, 2023-06-23:
@mniestroj @JordanYates FYI

nkaufmn, 2023-06-23:
In my initialisation of lorawan I have this code snippet:

```
struct lorawan_downlink_cb downlink_cb = {
.port = LW_RECV_PORT_ANY,
.cb = dl_callback
};

lorawan_register_downlink_callback(&downlink_cb);
```
 and the dl_callback is defined in the same file as:

```
static void dl_callback(uint8_t port, bool data_pending, int16_t rssi, int8_t snr, uint8_t len, const uint8_t *data)
{
	printk("Port %d, Pending %d, RSSI %ddB, SNR %ddBm\n", port, data_pending, rssi, snr);
	printk("Payload: ");
	if (data) {
		for(int i = 0; i < len; i++){
			printk("%d", data);
		}
	}
	printk("\n");
}
```

mniestroj, 2023-06-29:
> It only runs when everything is placed in one subprogramm, why can't I use more than one subprogram to access lorawan?

Could you elaborate? What do you mean by subprogram? Like using LoRaWAN APIs from two separate threads or what?

craigpeacock, 2023-07-17:
I appear to be having a similar issue. I get the following run-time error:

```
[00:00:23.037,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:00:23.037,000] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:00:23.037,000] <dbg> lorawan: mcps_confirm_handler: McpsRequest success!
[00:00:46.788,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:00:46.788,000] <err> os: ***** BUS FAULT *****
[00:00:46.788,000] <err> os:   Precise data bus error
[00:00:46.788,000] <err> os:   BFAR Address: 0x11f3861e
[00:00:46.789,000] <err> os: r0/a1:  0x00000001  r1/a2:  0x00ff7c82  r2/a3:  0x593e22d4
[00:00:46.789,000] <err> os: r3/a4:  0x000004af r12/ip:  0x00000c80 r14/lr:  0x0800ecdf
[00:00:46.789,000] <err> os:  xpsr:  0x21000000
[00:00:46.789,000] <err> os: Faulting instruction address (r15/pc): 0x08006142
[00:00:46.789,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:46.789,000] <err> os: Current thread: 0x20001170 (unknown)
[00:00:46.859,000] <err> os: Halting system
```

I moved all the LoRaWAN functions into subroutines called from main. 

What I have found through trial and error is  

```
	struct lorawan_downlink_cb downlink_cb = {
		.port = LW_RECV_PORT_ANY,
		.cb = &dl_callback
	};

	lorawan_register_downlink_callback(&downlink_cb);
```
has to be called from main. All the other functions can be put into subroutines.

Is it something like lorawan_register_downlink_callback() references the structure, not copies it and the compiler destroys the structure once the routine has been called.

JordanYates, 2023-07-17:
Almost certainly your problem is that your callback struct is being declared on the stack, and disappearing when the function returns. For example doing this WILL cause hard fault the next time a callback is generated:
```
void setup(void) {
	struct lorawan_downlink_cb downlink_cb = {
		.port = LW_RECV_PORT_ANY,
		.cb = &dl_callback
	};

	lorawan_register_downlink_callback(&downlink_cb);
}
```
The LoRaWAN subsystem requires that the provided struct remain valid for the lifetime of the program. Some ways to achieve that are:
1. Declare the struct in a function that never returns (your `main` suggestion above)
2. Declare the struct with the `static` qualifier, which will ensure it has a global lifetime
3. Declare the struct as a global variable

jhedberg, 2023-09-26:
> Yes it solved the problem. Thank you.

Thanks! Closing this then.

closed 2023-09-26 by jhedberg