zephyrproject-rtos/zephyr #59653: Bluetooth: Host: PAwR sync responder address is always identity address

hermabe, 2023-06-23:
**Describe the bug**
When a connection is established to a PAwR sync via the Periodic Advertising Connection Procedure, the responder address is always set to the id addr.
https://github.com/zephyrproject-rtos/zephyr/blob/bb5f421ade4cb3d880e2bd38bdbcaf849b9ce21b/subsys/bluetooth/host/hci_core.c#L1515-L1516

This is not correct if an RPA is used in the connection establishment. The advertiser/central does not agree with the sync/peripheral about what the responder address is. This leads to for example pairing not working.

**To Reproduce**
With `CONFIG_BT_PRIVACY=y`:
1. Set up a periodic advertiser with responses
2. Sync to it
3. Establish a connection using the Periodic Advertising Connection Procedure
4. Inspect the initiator and responder addresses
Or:
6. Initiate pairing
7. See that the DHKeyCheck fails because the central and peripheral has derived different keys from the addresses.

**Expected behavior**
The central and peripheral to agree on the initiator and responder addresses in all cases.

**Impact**
Bluetooth features requiring pairing are not available over a connection established from PAwR if a non-id address was used.

alwa-nordic, 2023-07-10:
> 3. Establish a connection using the Periodic Advertising Connection Procedure

I think you need to specify which address to connect to in this step. Does this problem appear if you connect to the identity address?

hermabe, 2023-07-17:
> > 3. Establish a connection using the Periodic Advertising Connection Procedure
>
> I think you need to specify which address to connect to in this step.

The address should be the one stored in the controller on the sync side (ie. the random or public address that is set over HCI).

> Does this problem appear if you connect to the identity address?

That is the one case where the identity address is the correct responder address.

carlescufi, 2023-08-03:
@hermabe we merged https://github.com/zephyrproject-rtos/zephyr/pull/57668 to allow you to make progress, is this issue still relevant?

hermabe, 2023-08-03:
> @hermabe we merged #57668 to allow you to make progress, is this issue still relevant?

It should be correct in the cases I have come across, but I am not confident it is correct in all configurations. I will have to sit down and figure it out, but I have not done that yet.

hermabe, 2023-12-04:
There is an open errata about how privacy for PAwR connections shall work. Will implement that once it is resolved.

alwa-nordic, 2023-12-04:
@hermabe Do you have a link to the erratum?

closed 2024-02-18 by github-actions[bot]