zephyrproject-rtos/zephyr #59664: Bluetooth: Controller: assert while scanning

Thalley, 2023-06-23:
**Describe the bug**
Using the shell to do `bt connect` which starts scanning, the following ASSERT were hit:
```
ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan.c:1196
```

**To Reproduce**
1. Do `bt connect` when using the BT shell
2. See error

May not be easy to reproduce as the issue was found in a very noisy environment. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
None

**Logs and console output**N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/59107

Using the nRF5340 with the hci_rpmsg sample with nrf5340_cpunet_iso-bt_ll_sw_split.conf on the netcore and the audio shell on the appcore.

**Additional context**
Found during UPF, so may be caused by significant traffic

cvinayak, 2023-10-03:
This is very different to reproduce (and do not see any other similar assertion encountered/reported). This issue should be discussed in Thursdays Zephyr Bluetooth meeting, and the priority can be changed to low.

cvinayak, 2024-01-12:
Marked as known issue.

Controller needs enhancement to reduce CPU usage overheads scheduling closely between radio events.

cvinayak, 2024-02-13:
Setting this priority low as the issue seen is old now.

Thalley, 2024-09-11:
@cvinayak it's been a while since this issue is created. Has anything changed?

cvinayak, 2024-09-11:
The said assertion has not been observed in the limited local testing or has been reported by users. We can close it as not planned.

Thalley, 2024-09-11:
> The said assertion has not been observed in the limited local testing or has been reported by users. We can close it as not planned.

Let's do that an re-open if seen again

closed 2024-09-11 by Thalley