zephyrproject-rtos/zephyr #59667: Ram Function section can be relocated with an offset

sdesfar, 2023-06-23:
**Describe the bug**

From ramfunc.ld:
```
SECTION_DATA_PROLOGUE(.ramfunc,,)
{
	MPU_ALIGN(__ramfunc_size);
	__ramfunc_start = .;
	*(.ramfunc)
	*(".ramfunc.*")

#include <snippets-ramfunc-section.ld>

	MPU_ALIGN(__ramfunc_size);
	__ramfunc_end = .;
} GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)
__ramfunc_size = __ramfunc_end - __ramfunc_start;
__ramfunc_load_start = LOADADDR(.ramfunc);
```

From xip.c:

```
z_early_memcpy(&__ramfunc_start, &__ramfunc_load_start, (uintptr_t) &__ramfunc_size);
```

The objective of this function is to relocate the whole section which is saved in Flash to the RAM, for further use.
Here, in case of effective alignment done by the MPU_ALIGN, the beginning of the section contains padding. This padding will be copied to the __ramfunc_start address, but according to the image of this section which is in flash, this address should contain the address of the first function. The copied data expected addresses do not match any more they real address.

I made a workaround by adding a SUBALIGN directive at the section entry:
```
SECTION_DATA_PROLOGUE(.ramfunc,,SUBALIGN(_region_min_align))
```

But I am not sure this is the right fix. I tried to move the `MPU_ALIGN(__ramfunc_size);` after the `__ramfunc_start = .;` but without success.

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare an Arm image for Cortex M33 with the following parameters:
   a. CONFIG_ARCH_HAS_RAMFUNC_SUPPORT=y
   b. CONFIG_XIP=y 
   c. CONFIG_MPU=y
   d. CONFIG_ARM_MPU=y
3. Prepare a XIP image, with some functions to be relocated in RAM (using the `__ramfunc` directive).
4. Make sure that when the linker will execute the ramfunc.ld, "." is not aligned with the MPU alignment and the `MPU_ALIGN(__ramfunc_size);` directevice will effectively have an impact on "."
5. Execute the binary, and break on the Ram function. The effectively executed code does not corresponds to the one in the flash. There is an offset due to the alignment.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Immediate crash when calling a function from the RAM. Can happen or not depending on the previous alignment of previous sections.

**Environment (please complete the following information):**
 - OS: Fedora 38
 - Toolchain: gcc-arm-none-eabi-9-2020-q2-update from arm website
 - Version: v3.1.0, looks also present on latest.

carlocaione, 2023-06-23:
Yes, using `SUBALIGN` can be a correct fix. @sdesfar do you want to open a PR or do you want me to open that for you?

sdesfar, 2023-06-29:
Hello @carlocaione, sorry for the late answer. I though I would have time this week to provide a PR, but unfortunately no... So you can provide a patch on your own if you have time. Thanks!

raffi-g, 2023-11-14:
good morning

We have been using the fix with `SUBALIGN` succesfully in our fork for a while now. However, we would like to get back to upstream zephyr again at some point. So I would open the PR with the fix, if that's still ok.

However, I came across #62368, which I believe is a duplicate of this issue. #62368 is blocked though, since they rather want to fix the root cause in #58080. Am I correct in assuming, that this also blocks this issue here?

closed 2024-01-29 by github-actions[bot]