zephyrproject-rtos/zephyr #59674: LE Audio: BAP: Shell: Add support for broadcast code as the broadcast sink

Thalley, 2023-06-23:
**Is your enhancement proposal related to a problem? Please describe.**
Add support for providing a broadcast code when syncing to a broadcast source as the broadcast sink. 

**Describe the solution you'd like**
Add a broadcast_code parameter to `cmd_sync_broadcast`

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

AayushGupta69, 2023-06-23:
@Thalley Can I work on this issue? I am new to open source and would like to contribute.

Thalley, 2023-06-23:
> @Thalley Can I work on this issue? I am new to open source and would like to contribute.

Absolutely! :) You can draw inspiration from the `bap_broadcast_assistant` shell module which does handle the broadcast code. I would suggest to have a look at one of my PRs that fixes an issue with it though (https://github.com/zephyrproject-rtos/zephyr/pull/59392). You should be able to more or less do something similar to that

AayushGupta69, 2023-06-24:
@Thalley Thanks for replying. I will start working on this.

Thalley, 2023-06-25:
> @Thalley Thanks for replying. I will start working on this.

Feel free to ping me here or on Discord if you encounter any issues

AayushGupta69, 2023-06-25:
Hi @Thalley, I apologize for unassigning myself from the GitHub issue. I have other commitments that require my attention. Sorry for any inconvenience caused.

Thalley, 2023-06-25:
> Hi @Thalley, I apologize for unassigning myself from the GitHub issue. I have other commitments that require my attention. Sorry for any inconvenience caused.

No problem - Let me know if you want to take it up later if it hasn't already been done by then

Thalley, 2024-01-09:
@Coalby Any progress on this?

closed 2024-02-28 by fabiobaltieri