zephyrproject-rtos/zephyr #59683: net: net_if_ipv6_addr_rm() always remove the solicited-node multicast address

vaussard, 2023-06-23:
**Describe the bug**

I am not sure if this is a real issue or just me doing something unexpected.

Here is my simplified setup (running on Zephyr 2.7 but `net_if_ipv6_addr_rm()` did not changed much):
- A Zephyr board has an IPv6 interface with a link-local address, for example `fe80::282e:a6ff:fe12:9270` based on its MAC address `2a:2e:a6:12:92:70`.
- Consequently it joins the multicast group `ff02::1:ff12:9270` on boot as shown by `net iface` on the console (see below).
- Now the board is connected to some router. Upon receiving a router advertisement, the interface is assigned a SLAAC address, for example `fd11:8963:1c1:7:282e:a6ff:fe12:9270`. Both addresses are in the same multicast group since they derive from the same MAC address.
- When disconnecting the link, we call `net_if_ipv6_addr_rm()` in order to remove the SLAAC address which is no longer usable. This is to prevent other issues if we are later connected to another router with a different prefix. This is maybe something that we are not expected to do, but other parts of the stack like `net_if_ipv6_dad_failed()` does the same, so I would expect a similar problem in that case.
- `net_if_ipv6_addr_rm()` always removes the solicited-node multicast address, without checking if it is used by other unicast addresses. We end up with a link-local address which is not part of the solicited-node multicast group, as shown by `net iface`. As a result, multicast packets (e.g. Neighbour Discovery Protocol) coming on this local link (when connected to a non-routing peer) will be dropped by `net_ipv6_input()` because we left the multicast group.

At boot:
```
Link addr : 2A:2E:A6:12:92:70
MTU       : 576
Flags     : AUTO_START,IPv6
IPv6 unicast addresses (max 2):
      fe80::282e:a6ff:fe12:9270 autoconf preferred infinite
IPv6 multicast addresses (max 4):
      ff02::1
      ff02::1:ff12:9270
      ff02::fb
IPv6 prefixes (max 2):
      <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 33210
IPv6 retransmit timer    : 0
```

After `net_if_ipv6_addr_rm()`:
```
Link addr : 2A:2E:A6:12:92:70
MTU       : 576
Flags     : AUTO_START,IPv6
IPv6 unicast addresses (max 2):
      fe80::282e:a6ff:fe12:9270 autoconf preferred infinite
IPv6 multicast addresses (max 4):
      ff02::1
      ff02::fb
IPv6 prefixes (max 2):
      <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 33210
IPv6 retransmit timer    : 0
```

I solved that issue in `net_if_ipv6_addr_rm()` by counting the number of unicast addresses which are sharing the multicast address, and skipping the `net_if_ipv6_maddr_rm()` if we have more than 1 user. Alternatively we could have some reference counting in `struct net_if_mcast_addr` instead of the `is_used` field, but this requires more changes all over the place.

I am happy to submit my fix if it seems suitable.

jukkar, 2023-06-24:
Your analysis sounds good, so please submit your fix.

vaussard, 2023-06-27:
Thank you for the feedback @jukkar. I submitted the fix in #59791.

commit 2eb7433f (2023-06-28): net: if: do not remove the solicited-node multicast address if used

Two different IPv6 addresses can have the same solicited node multicast
address, for example when they are derived from the same EUI-64
interface identifier during the auto-configuration process. For example,
an interface with a physical address 70:07:12:34:56:78 can have the
following:
- link-local address FE80::7207:12FF:FE34:5678
- global unicast address 2001:1234:::7207:12FF:FE34:5678

Both addresses will have the same solicited-node multicast address
FF02::1:FF34:5678. Currently, if one removes the global unicast address [... cut]
closed 2023-06-28 by fabiobaltieri