zephyrproject-rtos/zephyr #59718: TF-M does not generate correct MCUboot header

nandojve, 2023-06-23:
**Describe the bug**
The TF-M module uses MCUboot as default bootloader. In this case, the bootloader uses same version defined for Zephyr for compatibility reasons. Besides that, Zephyr build create the correct images for bootloader, S and NS firmwares. My guess is that the sign + merge procedures are not correct and final image is produced with invalid MCUboot headers. For some reason the image boot but the state on MCUboot is complete inconsistent.

**To Reproduce**

***Zephyr Mainline***

Apply the patch on trusted-firmware-m to print boot log
```diff
diff --git a/config/build_type/minsizerel.cmake b/config/build_type/minsizerel.cmake
index 3cedb580c..25117e144 100644
--- a/config/build_type/minsizerel.cmake
+++ b/config/build_type/minsizerel.cmake
@@ -5,7 +5,7 @@
 #
 #-------------------------------------------------------------------------------
 
-set(MCUBOOT_LOG_LEVEL                   "OFF"      CACHE STRING    "Level of logging to use for MCUboot [OFF, ERROR, WARNING, INFO, DEBUG]")
+set(MCUBOOT_LOG_LEVEL                   "INFO"      CACHE STRING    "Level of logging to use for MCUboot [OFF, ERROR, WARNING, INFO, DEBUG]")
 set(TFM_SPM_LOG_LEVEL                   TFM_SPM_LOG_LEVEL_SILENCE CACHE STRING "Set minsizerel SPM log level as Silence level")
 set(TFM_PARTITION_LOG_LEVEL             TFM_PARTITION_LOG_LEVEL_SILENCE CACHE STRING "Set minsizerel SP log level as Silence level")
 set(TFM_BL1_LOGGING                     OFF          CACHE BOOL      "Whether BL1 will log to uart")
```

How to build and flash
```
west build -b b_u585i_iot02a_ns samples/tfm_integration/psa_protected_storage
./build/tfm/regression.sh
west flash
```

Highlight problems
```diff
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Boot source: primary slot
-[INF] Swap type: none
-[INF] Primary image: magic=bad, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
-[INF] Boot source: none
-[INF] Swap type: none
[INF] Bootloader chainload address offset: 0x38000
[INF] Jumping to the first image slot
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[Sec Thread] Secure image initializing!
Booting TF-M v1.7.0+79a6115d3
Creating an empty ITS flash layout.
Creating an empty PS flash layout.
*** Booting Zephyr OS build zephyr-v3.4.0-382-g7584c1737495 ***
```

***trusted-firmware-m.git***

https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/

`mainline` a4f189c77cdd59321a7707a930d9adf3e9e46543

Apply the patch on trusted-firmware-m to print boot log, use only internal memory and disable STSAFE secure device. This makes the board exactly what is defined on Zephyr.
```diff
diff --git a/config/build_type/minsizerel.cmake b/config/build_type/minsizerel.cmake
index 3cedb580c..25117e144 100644
--- a/config/build_type/minsizerel.cmake
+++ b/config/build_type/minsizerel.cmake
@@ -5,7 +5,7 @@
 #
 #-------------------------------------------------------------------------------
 
-set(MCUBOOT_LOG_LEVEL                   "OFF"      CACHE STRING    "Level of logging to use for MCUboot [OFF, ERROR, WARNING, INFO, DEBUG]")
+set(MCUBOOT_LOG_LEVEL                   "INFO"      CACHE STRING    "Level of logging to use for MCUboot [OFF, ERROR, WARNING, INFO, DEBUG]")
 set(TFM_SPM_LOG_LEVEL                   TFM_SPM_LOG_LEVEL_SILENCE CACHE STRING "Set minsizerel SPM log level as Silence level")
 set(TFM_PARTITION_LOG_LEVEL             TFM_PARTITION_LOG_LEVEL_SILENCE CACHE STRING "Set minsizerel SP log level as Silence level")
 set(TFM_BL1_LOGGING                     OFF          CACHE BOOL      "Whether BL1 will log to uart")
diff --git a/platform/ext/target/stm/b_u585i_iot02a/config.cmake b/platform/ext/target/stm/b_u585i_iot02a/config.cmake
index 04cbb0797..301bad25f 100644
--- a/platform/ext/target/stm/b_u585i_iot02a/config.cmake
+++ b/platform/ext/target/stm/b_u585i_iot02a/config.cmake
@@ -32,5 +32,5 @@ set(MCUBOOT_FIH_PROFILE                 LOW         CACHE STRING    "Fault injec
 set(CONFIG_TFM_USE_TRUSTZONE             ON)
 set(TFM_MULTI_CORE_TOPOLOGY              OFF)
 set(PLATFORM_HAS_FIRMWARE_UPDATE_SUPPORT ON)
-set(STSAFEA                             ON          CACHE BOOL      "Activate ST SAFE SUPPORT")
+set(STSAFEA                             OFF          CACHE BOOL      "Activate ST SAFE SUPPORT")
 set(MCUBOOT_DATA_SHARING                ON)
diff --git a/platform/ext/target/stm/b_u585i_iot02a/include/flash_layout.h b/platform/ext/target/stm/b_u585i_iot02a/include/flash_layout.h
index 912f48feb..380eddf2e 100644
--- a/platform/ext/target/stm/b_u585i_iot02a/include/flash_layout.h
+++ b/platform/ext/target/stm/b_u585i_iot02a/include/flash_layout.h
@@ -20,7 +20,7 @@
 
 #define OSPI_FLASH_TOTAL_SIZE           (0x4000000)  /* 64 MB same as MX25LM51245G_FLASH_SIZE */
 #define OSPI_FLASH_BASE_ADDRESS         (0x70000000) /* same as OCTOSPI1_BASE  */
-#define EXTERNAL_FLASH
+//#define EXTERNAL_FLASH
 /* This header file is included from linker scatter file as well, where only a
  * limited C constructs are allowed. Therefore it is not possible to include
  * here the platform_retarget.h to access flash related defines. To resolve this
```

To build and flash without confirm the image (used for firmware upgrade)
```
mkdir build && cd build
cmake .. -DTFM_PLATFORM=stm/b_u585i_iot02a
cmake --build ./  -- install
./postbuild.sh
./regression.sh
./TFM_UPDATE.sh
```

Highlight correct expectations
```diff
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
+[INF] Boot source: primary slot
+[INF] Swap type: test
+[INF] Image upgrade secondary slot -> primary slot
+[INF] Erasing the primary slot
+[INF] Copying the secondary slot to the primary slot: 0xzx bytes
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
+[INF] Boot source: primary slot
+[INF] Swap type: test
+[INF] Image upgrade secondary slot -> primary slot
+[INF] Erasing the primary slot
+[INF] Copying the secondary slot to the primary slot: 0xzx bytes
[INF] Bootloader chainload address offset: 0x38000
[INF] Jumping to the first image slot
Non-Secure system starting...

```

To build and flash without confirm the image (used for production)
```
mkdir build && cd build
cmake .. -DTFM_PLATFORM=stm/b_u585i_iot02a -DMCUBOOT_CONFIRM_IMAGE=ON
cmake --build ./  -- install
./postbuild.sh
./regression.sh
./TFM_UPDATE.sh
```

Highlight correct expectations
```diff
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
+[INF] Boot source: primary slot
+[INF] Swap type: perm
+[INF] Image upgrade secondary slot -> primary slot
+[INF] Erasing the primary slot
+[INF] Copying the secondary slot to the primary slot: 0xzx bytes
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
+[INF] Boot source: primary slot
+[INF] Swap type: perm
+[INF] Image upgrade secondary slot -> primary slot
+[INF] Erasing the primary slot
+[INF] Copying the secondary slot to the primary slot: 0xzx bytes
[INF] Bootloader chainload address offset: 0x38000
[INF] Jumping to the first image slot
```

**Expected behavior**
The Zephyr build should produce images with correct signature and correct MCUboot headers.

**Impact**
This is a serious problem since TF-M + MCUboot are supposed to be safe and secure.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.1
 - Zephyr mainline 7584c173749534f70a42c5dc18353761188067a1
 - trusted-firmware-m 79a6115d3a8d0e04864ae8156c1dc8532b750f5a (from west)

**Additional context**
This can be verified on Zephyr 3.2 which means is not a new problem.

CC: @otavio

microbuilder, 2023-06-23:
Moving this to medium. High should be something affecting the kernel or with much wider impact, IMO, than TF-M which is still a relatively niche subsystem today.

It isn't clear to me what the problem is exactly, though. Are you saying that the signing script that is part of the TF-M/Zephyr build process isn't correctly signing images? Have you tested with a board other than the ST U585 to determine if this is isolated to that board, or also affects others, and if it isn't related to the post-build scripts ST has in place?

It looks like you are getting some warnings when you flash TF-M (S) + Zephyr (NS) images built and signed by the Zephyr build system, and flashed directly for the first time on a clean device. Have you tested with a second image (with a higher version number to get past rollback protection), passing that second image in for verification with something like the `mcumgr` tool, since we don't support the TF-M firmware update service (FWU) at this time?

With ZDS next week, there won't be much bandwidth from anyone to dig into this, but if you can do a bit of digging to look at the `imgtool` parameters in both cases and compare them, and maybe test on another board or trying to do an update, it will make it a lot easier to narrow down what's happening here. Thanks!

nandojve, 2023-06-24:
Hi @microbuilder ,

> It isn't clear to me what the problem is exactly, though. Are you saying that the signing script that is part of the TF-M/Zephyr build process isn't correctly signing images? Have you tested with a [...]

Something like that. As you can see by the logs from Zephyr image there is present an entry with `magic=bad`. I have a nrf5340dk_nrf5340 (2.0.1, 2023.10) board but I don't understand why the logs from MCUboot are not shown. Even when I build trusted-firmware-m on debug MCUboot logs are not present. Can you pass me a configuration that shows the log for this nRF board so I can help to isolate this issue?

> It looks like you are getting some warnings when you flash TF-M (S) + Zephyr (NS) images built and signed by the Zephyr build system, and flashed directly for the first time on a clean device. Have you tested with a second image (with a higher version number to get past rollback protection), passing that second image in for verification with something like the mcumgr image tool, since we don't support the TF-M firmware update service (FWU) at this time?

I didn't understand this comment. Are you referring to what messages exactly?

BTW, you need to take in consideration that there is no connection with FWU here. It is a simple build from any example from mainline and you get this state on MCUboot image using U585. On the other hand, this does not happen when I build directly from trusted-firmware-m.

As an additional info, I tested and run trusted-firmware-m mainline, release-1.8.x and release-1.7.x. I did not found issues on ST U585 board.

microbuilder, 2023-06-25:
> I didn't understand this comment. Are you referring to what messages exactly?

What I meant was have you tried doing a firmware update with `mcumgr` from the command line, since there is no FWU service available at this time, and see what log messages you get after the update?

Also, if you have time to look through the build logs and list/compare the `imgtool` parameters being passed to sign the images in the Zephyr and TF-M cases, it would be useful to see if TF-M is doing something we're not when building and signing with Zephyr. I won't have much debug bandwidth myself until after ZDS this week, sorry.

microbuilder, 2023-07-26:
@nandojve were you able to do any further research on this, to try to give @d3zd3z some help determining if this behavior is normal?

I've recently change companies and have handed TF-M integration maintainership over to David, but can you follow up with the previous questions to give him some feedback since he has limited time himself? Thanks!

nandojve, 2023-07-27:
Hi @microbuilder ,

No, unfortunately I didn't dig to dip on this since ZDS. My last test was using #60226 but results are the same. As I already mentioned, I have a [b_u585i_iot02a](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/tree/platform/ext/target/stm/b_u585i_iot02a) and a [nrf5340dk_nrf5340_cpuapp](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/tree/platform/ext/target/nordic_nrf/nrf5340dk_nrf5340_cpuapp) to try and test.

I was stuck with nrf5340dk because I didn't found a way to enable MCUboot logs. So, I couldn't compare results from Zephyr TFM and the mainline baremetal TFM itself. Even using the TFM baremetal I did not managed to have the MCUboot logs enabled.

> It looks like you are getting some warnings when you flash TF-M (S) + Zephyr (NS) images built and signed by the Zephyr build system, and flashed directly for the first time on a clean device. 

This seems to be normal for U5 which, on the first boot, update something. This happens even with the baremetal TFM delivered directly from ST using v1.3. My guess is that this may not be the issue because on TFM baremetal mainline this happens too and the system works as expected.

I have this and next week which I could spend more on this with some help.

joerchan, 2023-11-28:
> I was stuck with nrf5340dk because I didn't found a way to enable MCUboot logs. So, I couldn't compare results from Zephyr TFM and the mainline baremetal TFM itself. Even using the TFM baremetal I did not managed to have the MCUboot logs enabled.

@nandojve The BL2 logs (TF-Ms mcuboot variant) is using UART1 to log output, not UART0 as the application is doing.

Build and running:
```
west build -p -b nrf5340dk_nrf5340_cpuapp_ns samples/tfm_integration/psa_protected_storage
west flash --erase
```

Secure VCOM (ttyACM0)
```
[INF] Starting bootloader
[INF] Beginning BL2 provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[INF] Image index: 1, Swap type: none
[INF] Image index: 0, Swap type: none
[INF] Bootloader chainload address offset: 0x10000
[INF] Jumping to the first image slot
[INF] Beginning TF-M provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[Sec Thread] Secure image initializing!
Booting TF-M v1.5.0+33c0f47bc
Creating an empty ITS flash layout.
Creating an empty PS flash layout.
[INF][Crypto] Provisioning entropy seed... complete.
```
Non-secure VCOM (ttyACM1)
```
*** Booting Zephyr OS build zephyr-v3.5.0-1996-g1ff79e4825d3 ***
Protected Storage sample started.
PSA Protected Storage API Version 1.0
Writing data to UID1: The quick byog
Info on data stored in UID1:
- Size: 44
- Capacity: 0x 1
- Flags: 0x 0                                                              
Read and compare data stored in UID1                                       
Data stored in UID1: The quick brown fox jumps over thg data stored in UID1: Lorem ipsum dolor sit amet
Writing data to UID2 with overwrite protection: The quick brown fox jumps over the lazy dog
Attempting to write 'Lorem ipsum dolor sit amet' to UID2                   
Got expected error (PSA_ERROR_NOT_PERMITTED) when writing to protected UID
Removing UID1
```
I'm using the 2.0.0 revision of the board. If you have the 1.x.x board you may have to use some wires to get the secure output.
For details on the Secure output see here:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/device_guides/working_with_nrf/nrf53/nrf5340.html#getting-logging-output

closed 2024-04-12 by github-actions[bot]