zephyrproject-rtos/zephyr #59752: Device crashed due to bus fault while executing null work->handler in work_queue_main()

AshwiniMShinde-eaton, 2023-06-26:
I've built webserver application using civetweb and running it on wifi interface(using infineon WHD) that uses zephyr network stack.
Zephyr version - 3.2.0
Hardware - STM32U575ZI_Q connected to wifi Murata 1DX module over SDIO interface
When I run my pytest suite that sends REST request consistently. It runs test cases where it opens TCP connection with the device, executes REST URI and closes TCP connection. This process repeats for all the test cases and at some point device crashes due to bus fault when work_queue_main() executes null work->handler. Following is the callstack when the device crashes:

[image]

It seems that something in tcp_workq causes this null handler scenario but I'm unable to find the exact root cause. Temporarily I've added the following workaround in work.c file just to bypass this null handler scenario(which is not good practice)
[image]

I have the following questions:

1. Can you provide some pointers on what could be the root cause or how to debug tcp_workq scenario that leads to null work->handler?
2. I understand that the workaround that I posted above is not recommended. This workaround helps us to avoid the device crash but we're not aware what impact it will have. Could you provide your inputs here?

nordicjm, 2023-06-26:
Just and observation from your log that might indicate you have a bit of (kernel?) memory corruption, a *valid* HTTP header first line would end in something like HTTP/1.1, you have garbage after the slash in your debug output though

jhedberg, 2023-07-11:
Can you reproduce this with the latest Zephyr version?

jhedberg, 2023-07-11:
Cc @rlubos

carlescufi, 2023-07-12:
> It seems that something in tcp_workq causes this null handler scenario but I'm unable to find the exact root cause. Temporarily I've added the following workaround in work.c file just to bypass this null handler scenario(which is not good practice)

This looks like a stack overflow in the TCP workqueue, try increasing [this value](https://github.com/zephyrproject-rtos/zephyr/blob/1601725354e0e099d3b3d90f334bcf49a9300fdf/subsys/net/ip/Kconfig#L509) to see if it fixes it.

AshwiniMShinde-eaton, 2023-07-12:
I tried to increase CONFIG_NET_TCP_WORKQ_STACK_SIZE and test. This reduces the probability of issue but doesn't completely fix it. The only difference this change makes is the device crashes little late. For example, if the device crashes after 15-20mins in normal cases then after increasing CONFIG_NET_TCP_WORKQ_STACK_SIZE the device will crash after an hour or so

rlubos, 2023-07-12:
> It seems that something in tcp_workq causes this null handler scenario but I'm unable to find the exact root cause. 

NULL work `handler` clearly seems like some memory corruption, but it doesn't have to be anything in the `tcp_workq`, it could be some other part of the code that causes the corruption. Such issues are tricky to identify and practically impossible to find w/o the application on the table with the debugger attached.

I suggest to enable STACK_SENTINEL and asserts and try to dig further.

carlescufi, 2023-07-26:
@AshwiniMShinde-eaton any news on this? could the issue be in your own code?

AshwiniMShinde-eaton, 2023-07-26:
> @AshwiniMShinde-eaton any news on this? could the issue be in your own code?

We're working on it. We're trying to upgrade our wifi stack and test the same. We'll get back to you once we have some findings there

carlescufi, 2023-10-04:
> > @AshwiniMShinde-eaton any news on this? could the issue be in your own code?
> 
> We're working on it. We're trying to upgrade our wifi stack and test the same. We'll get back to you once we have some findings there

@AshwiniMShinde-eaton please update us on this issue.

AshwiniMShinde-eaton, 2023-10-07:
> > > @AshwiniMShinde-eaton any news on this? could the issue be in your own code?
> > 
> > 
> > We're working on it. We're trying to upgrade our wifi stack and test the same. We'll get back to you once we have some findings there
> 
> @AshwiniMShinde-eaton please update us on this issue.

Sorry we didn't get chance to revisit this again as we were occupied with some other work. Will get back to you once we switch to working on this.

talih0, 2023-11-19:
I saw a similar problem in TCP work. It also happened right after the socket was closed.

```[00:12:38.227,000] <err> os:   Illegal use of the EPSR```                                                                                                                                                                                                        
```[00:12:38.228,000] <err> os: r0/a1:  0x20018ef8  r1/a2:  0xe000ed00  r2/a3:  0x00000000```                                                                                                                                                                      
```[00:12:38.228,000] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x08046119```                                                                                                                                                                       
```[00:12:38.228,000] <err> os:  xpsr:  0x60000000```                                                                                
```[00:12:38.228,000] <err> os: Faulting instruction address (r15/pc): 0x00000000```                                                                                                                                                                                
```[00:12:38.228,000] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0```                                                                                                                                                                                
```[00:12:38.229,000] <err> os: Current thread: 0x1fff0108 (tcp_work)```

erwango, 2023-11-20:
> I saw a similar problem in TCP work. It also happened right after the socket was closed.

Did you try [this proposal](https://github.com/zephyrproject-rtos/zephyr/issues/59752#issuecomment-1632150448) ?

talih0, 2023-11-20:
> > I saw a similar problem in TCP work. It also happened right after the socket was closed.
> 
> Did you try [this proposal](https://github.com/zephyrproject-rtos/zephyr/issues/59752#issuecomment-1632150448) ?

Not yet, I'll give it a try.

paussems, 2023-11-23:
Hello, I have this crash too and did some investigations which led me to understand the problem.

The crash happens when `tcp_resend_data` is called by the `send_data_timer` with the following values in the connection structure:
- in_retransmission = true
- send_data_retries < CONFIG_NET_TCP_RETRY_COUNT
- send_retries = 0
 
In that case we have the following chain of execution in `tcp_resend_data`: `tcp_send_data()` -> `tcp_out_ext()` -> `tcp_send_process_no_lock()` -> `tcp_conn_close()`, which causes the `conn` structure to be blanked by a `memset(conn, 0x00)` call.

The problem is that after calling `tcp_send_data()`, the `tcp_resend_data` function resumes its execution and keeps using the `conn` struct like it was still valid.

What causes the fatal crash is the call to `k_work_reschedule_for_queue(send_data_timer)`: at expiration of the timer the `conn->send_data_timer.work.handler` which has been set to `NULL` is called, and kaboom ;-)

I'm able to reproduce this issue in less than 30s when doing socket connect / disconnect in a loop on a specific service which is behind a load balancer, not sure why it affects the working of the TCP stack, but the good thing is that I can test proposed fixes.

rlubos, 2023-11-27:
Thanks @paussems for the detailed analysis. The issue seems quite tricky to reproduce, I've tried hard today but with no success. But looking at the code, based on the information you provide, it's hard to deny that the problem exists (we should definitely avoid dereferencing the TCP connection while it can still be used from the same context).

Thinking about the solution for this issue, I think that it'd be reasonable to move the actual TCP context deallocation to a separate work item. As @talih0 suggested in some other thread, that should also prevent any potential race with any other scheduled work item, as we'll be 100% certain then, that any other pending work will be removed from the queue when canceled. I'll try send a fix proposal soon.

paussems, 2023-11-27:
Hi @rlubos , I have a minimal project for ESP32 which reproduces the problem on https://gitlab.com/esnah/zephyr-59752

rlubos, 2023-11-28:
@paussems Thanks for the additional input. I didn't manage to reproduce **exactly** the issue, but with some ugly tweaks with artificial packet loss and bogus delays in the TCP work handlers I was able to finally encounter the race, so I had something to verify the results.

I've submitted https://github.com/zephyrproject-rtos/zephyr/pull/65888 with the solution described earlier, we now submit a dedicated work to release the TCP context, to avoid race with other work handlers. Seems to help nicely in the test scenario I've created. Please give it a try and tell if it solves the issues you observe as well.

paussems, 2023-11-28:
> I've submitted #65888 with the solution described earlier, we now submit a dedicated work to release the TCP context, to avoid race with other work handlers. Seems to help nicely in the test scenario I've created. Please give it a try and tell if it solves the issues you observe as well.

Thanks @rlubos for looking into this issue so quickly. I'm running the test program with #65888 applied as we speak, and it's still running fine after a couple of minutes, whereas it crashed within 15s just before I applied the patches.

I have some errors once in a while in my logs, but they do not make the device crash.
```
[00:09:39.128,000] <inf> net_websocket_client_sample: ...
[00:09:40.368,000] <err> net_tcp: net_pkt alloc failure
[00:09:40.493,000] <err> net_tcp: conn: 0x3fffaa7c packet allocation failed, len=53
[00:09:40.493,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:09:40.698,000] <err> net_tcp: conn: 0x3fffacd4 packet allocation failed, len=53
[00:09:40.698,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:09:40.804,000] <err> net_tcp: conn: 0x3fffaf2c packet allocation failed, len=53
[00:09:40.804,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:09:43.368,000] <err> net_websocket_client_sample: connect: -1
[00:09:43.369,000] <inf> net_websocket_client_sample: ...
```

 I will leave it running overnight and I'll report tomorrow morning (CET TZ), but so far it's looking very good !

paussems, 2023-11-29:
Hello again @rlubos, the device is still running with #65888 after 15h of continuous connects/disconnects on the setup where I could reproduce the issue, so on my side it's confirmed that your new release implementation fixes the race condition.

Have a good day!

rlubos, 2023-11-29:
Excellent @paussems, thank you

closed 2023-11-30 by carlescufi