zephyrproject-rtos/zephyr #59776: stm32f756zg

andy9a9, 2023-06-27:
Dear community!

Would it be possible to add support for 1MB flash on [NUCLEO-F756ZG
](https://www.st.com/en/evaluation-tools/nucleo-f756zg.html)? While there is no support, it's not possible to use an update feature via mcuboot.

I tried to do it by myself, but didn't work like I'm expecting.
```
&flash0 {

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/*
		 * The two sectors 0-1 (32+32 kbytes) are reserved for
		 * the bootloader.
		 */
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x0 DT_SIZE_K(64)>;
			read-only;
		};;

		/*
		 * The flash starting at 0x00010000 and ending at
		 * 0x0001ffff (sectors 16-31) is reserved for use
		 * by the application.
		 */
		storage_partition: partition@1e000 {
			label = "storage";
			reg = <0x0001e000 0x00002000>;
		};

		slot0_partition: partition@20000 {
			label = "image-0";
			reg = <0x00020000 0x00060000>;
		};
		slot1_partition: partition@80000 {
			label = "image-1";
			reg = <0x00080000 0x00060000>;
		};
		scratch_partition: partition@e0000 {
			label = "image-scratch";
			reg = <0x000e0000 0x00020000>;
		};
	};
};
```
After that boards booted, but there is a message `Cannot upgrade: slots have non-compatible sectors`... .
```
uart:~$ *** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
I: Starting bootloader
W: Cannot upgrade: slots have non-compatible sectors
I: Bootloader chainload address offset: 0x20000
�: Jumping to the first image slot

*** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
[00:00:00.011,000] <inf> main: Hawkbit sample app started
[00:00:00.011,000] <inf> main: Image build time: Jun 26 2023 15:19:
```
At the end, I was trying to get the `Hawkbit` client running, but it didn't init the network at all. 
So what I'm doing wrong?

Thanks

FRASTM, 2023-06-27:
Should the flash partition be aligned on sector :
```
The Flash memory is organized as follows:
• On STM32F756xx and STM32F74xx devices, a main memory block divided into 4 
sectors of 32 Kbytes, 1 sector of 128 Kbytes and 3 sectors of 256 Kbytes
```

andy9a9, 2023-06-27:
> Should the flash partition be aligned on sectors :
> 
> ```
> The Flash memory is organized as follows:
> • On STM32F756xx and STM32F74xx devices, a main memory block divided into 4 
> sectors of 32 Kbytes, 1 sector of 128 Kbytes and 3 sectors of 256 Kbytes
> ```

OK, I changed it according to your suggestion.
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* 128KB for bootloader */
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(128)>;
			read-only;
		};

		/* storage: 256KB for settings */
		storage_partition: partition@20000 {
			label = "storage";
			reg = <0x00020000 DT_SIZE_K(256)>;
		};

		/* application image slot: 256KB */
		slot0_partition: partition@60000 {
			label = "image-0";
			reg = <0x00060000 DT_SIZE_K(256)>;
		};

		/* backup slot: 256KB */
		slot1_partition: partition@A0000 {
			label = "image-1";
			reg = <0x000A0000 DT_SIZE_K(256)>;
		};
	};
};
```
I think, that I did it wrongly, because it's worst as it was before.
```
*** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
E: Image in the primary slot is not valid!
E: Unable to find bootable image
```

FRASTM, 2023-06-27:
It also depends on what is in chosen as `zephyr,code-partition = &slot0_partition;`
Here it seems that no image in slot0 or 1 has the correct header

andy9a9, 2023-06-28:
> It also depends on what is in chosen as `zephyr,code-partition = &slot0_partition;` Here it seems that none of the images stored in slot0 or 1 has the correct header like : [image]

Yes, I choose `zephyr,code-partition = &slot0_partition`
```
	chosen {
		zephyr,console = &usart3;
		zephyr,shell-uart = &usart3;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &slot0_partition;
		zephyr,dtcm = &dtcm;
```
How to read/verify, that the header is wrong? Or do you have a sample project, which can I test?

FRASTM, 2023-06-28:
Header of the image comes with the build command (--sysbuild _?_)
You can examine (gdb or stm32cubeprogrammer) the content of the flash memory at the address of the image and see that this kind of header is present (at offset 0 is the header then at offset 0x400 starts the code)

- Another point to check if that the `west flash` commands does not erase the image that you have just programmed when it writes the mcuboot  (see the https://github.com/zephyrproject-rtos/zephyr/issues/47781)

andy9a9, 2023-06-29:
I try to build the `hawkbit` with `--sysbuild`, but it failed:
```
west build -b nucleo_f756zg --sysbuild

[4/321] Generating include/generated/version.h
-- Zephyr version: 3.4.0-rc1 (/home/vboxuser/zephyrproject/zephyr), build: v3.4.0-rc1-90-g5c7d2f1a90a6
[311/321] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/dfu/boot/libsubsys__dfu__boot.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/net/lib/http/libsubsys__net__lib__http.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/ethernet/libdrivers__ethernet.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  modules/mcuboot/libmcuboot_util.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -lm  -Wl,-lc  -L"/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m/nofp  -Wl,-lgcc  -lc && cd /home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit/zephyr && /usr/bin/cmake -E echo
/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `main':
/home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/src/main.c:37: undefined reference to `hawkbit_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FAILED: hawkbit-prefix/src/hawkbit-stamp/hawkbit-build /home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit-prefix/src/hawkbit-stamp/hawkbit-build 
cd /home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/hawkbit && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
```
But if I build it without it, it works fine:
```
west build -b nucleo_f756zg

[1/329] Preparing syscall dependency handling

[2/329] Generating include/generated/version.h
-- Zephyr version: 3.4.0-rc1 (/home/vboxuser/zephyrproject/zephyr), build: v3.4.0-rc1-90-g5c7d2f1a90a6
[319/329] Linking C executable zephyr/zephyr_pre0.elf

[323/329] Linking C executable zephyr/zephyr_pre1.elf

[329/329] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      181912 B       256 KB     69.39%
             RAM:       67328 B       256 KB     25.68%
            QSPI:          0 GB       256 MB      0.00%
            DTCM:       12544 B        64 KB     19.14%
        IDT_LIST:          0 GB         2 KB      0.00%
```
Then I signed the application via `west sign -t imgtool -- --key ../../../../../bootloader/mcuboot/root-rsa-2048.pem` and flash the bin/hex `west flash --bin-file ./build/zephyr/zephyr.signed.bin`,`west flash --hex-file ./build/zephyr/zephyr.signed.hex`.
```
west flash --hex-file ./build/zephyr/zephyr.signed.hex 
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: build/zephyr/zephyr.signed.hex
Open On-Chip Debugger 0.11.0+dev-00726-ge2b6f5655 (2023-05-11-16:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[... cut ...]
Info : device id = 0x10016449
Info : flash size = 1024 kbytes
Warn : Adding extra erase range, 0x08040000 .. 0x0805ffff
auto erase enabled
wrote 393216 bytes from file build/zephyr/zephyr.signed.hex in 11.020806s (34.843 KiB/s)

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoke
```
I looked at the memory via `STM32CubeProgrammer`. The memory starts at `0x08000000` and the slot offset is `0x00060000` => `0x8060000`.
[image]

FRASTM, 2023-06-29:
The offset  0x60000 is the address of the slot0_partition chosen by the mcuboot (primary image)

andy9a9, 2023-06-29:
> The offset 0x60000 is the address of the slot0_partition chosen by the mcuboot (primary image)

Ok, so what does it mean for me? Or what's wrong/should I do?

Laczen, 2023-06-30:
@andy9a9, the partitioning is still wrong, it is not aligned to the flash block sizes. The start of slot 0 is in the middle of a block. There also is no scratch partition.

You can also remove the storage partition as there is no storage solution for this flash block layout.

andy9a9, 2023-06-30:
@Laczen OK, but this configuration is still not working, while you have duplicated `slot1_partition`. If I removed it, then the compilation is failing due to missing `storage_partition`.
```
[142/329] Building C object zephyr/sub...ubsys__mgmt__hawkbit.dir/hawkbit.c.obj
FAILED: zephyr/subsys/mgmt/hawkbit/CMakeFiles/subsys__mgmt__hawkbit.dir/hawkbit.c.obj 
ccache /opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM7 -DHSE_VALUE=8000000 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DSTM32F756xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_ANSI_SOURCE -D_FORTIFY_SOURCE=1 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/vboxuser/zephyrproject/zephyr/include -I/home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/zephyr/include/generated -I/home/vboxuser/zephyrproject/zephyr/soc/arm/st_stm32/stm32f7 -I/home/vboxuser/zephyrproject/zephyr/lib/libc/newlib/include -I/home/vboxuser/zephyrproject/zephyr/lib/posix/getopt/. -I/home/vboxuser/zephyrproject/zephyr/drivers -I/home/vboxuser/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/vboxuser/zephyrproject/zephyr/subsys/net/l2 -I/home/vboxuser/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/vboxuser/zephyrproject/zephyr/subsys/net/lib/dns/. -I/home/vboxuser/zephyrproject/zephyr/subsys/net/ip -I/home/vboxuser/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/vboxuser/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/soc -I/home/vboxuser/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include -I/home/vboxuser/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include/Legacy -I/home/vboxuser/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I/home/vboxuser/zephyrproject/modules/crypto/tinycrypt/lib/include -I/home/vboxuser/zephyrproject/modules/crypto/mbedtls/include -I/home/vboxuser/zephyrproject/zephyr/modules/mbedtls/configs -I/home/vboxuser/zephyrproject/zephyr/modules/mbedtls/include -fno-strict-aliasing -Os -imacros /home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/vboxuser/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/vboxuser/zephyrproject/zephyr/samples/subsys/mgmt/hawkbit=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/vboxuser/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/vboxuser/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -D_POSIX_THREADS -std=c99 -MD -MT zephyr/subsys/mgmt/hawkbit/CMakeFiles/subsys__mgmt__hawkbit.dir/hawkbit.c.obj -MF zephyr/subsys/mgmt/hawkbit/CMakeFiles/subsys__mgmt__hawkbit.dir/hawkbit.c.obj.d -o zephyr/subsys/mgmt/hawkbit/CMakeFiles/subsys__mgmt__hawkbit.dir/hawkbit.c.obj -c /home/vboxuser/zephyrproject/zephyr/subsys/mgmt/hawkbit/hawkbit.c
In file included from /home/vboxuser/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:92,
                 from /home/vboxuser/zephyrproject/zephyr/include/zephyr/toolchain.h:50,
                 from /home/vboxuser/zephyrproject/zephyr/include/zephyr/sys/__assert.h:11,
                 from /home/vboxuser/zephyrproject/zephyr/include/zephyr/sys/iterable_sections.h:10,
                 from /home/vboxuser/zephyrproject/zephyr/include/zephyr/logging/log_instance.h:10,
                 from /home/vboxuser/zephyrproject/zephyr/include/zephyr/logging/log.h:10,
                 from /home/vboxuser/zephyrproject/zephyr/subsys/mgmt/hawkbit/hawkbit.c:11:
/home/vboxuser/zephyrproject/zephyr/subsys/mgmt/hawkbit/hawkbit.c: In function 'hawkbit_init':
/home/vboxuser/zephyrproject/zephyr/include/zephyr/device.h:85:41: error: '__device_dts_ord_DT_N_NODELABEL_storage_partition_PARENT_PARENT_ORD' undeclared (first use in this function)
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/home/vboxuser/zephyrproject/zephyr/include/zephyr/toolchain/common.h:132:26: note: in definition of macro '_DO_CONCAT'
  132 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
/home/vboxuser/zephyrproject/zephyr/include/zephyr/device.h:85:33: note: in expansion of macro '_CONCAT'
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/home/vboxuser/zephyrproject/zephyr/include/zephyr/device.h:211:37: note: in expansion of macro 'DEVICE_NAME_GET'
  211 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/home/vboxuser/zephyrproject/zephyr/include/zephyr/device.h:228:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  228 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/home/vboxuser/zephyrproject/zephyr/include/zephyr/storage/flash_map.h:344:9: note: in expansion of macro 'DEVICE_DT_GET'
  344 |         DEVICE_DT_GET(DT_MTD_FROM_FIXED_PARTITION(DT_NODELABEL(label)))
      |         ^~~~~~~~~~~~~
[... cut ...]
		slot1_partition: partition@C0000 {
			label = "image-1";
			reg = <0x000C0000 DT_SIZE_K(256)>;
		};
	};
};
```
I have the same results as before
```
*** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
E: Image in the primary slot is not valid!
E: Unable to find bootable imag
```

Laczen, 2023-06-30:
@andy9a9, if hawkbit requires a storage partition this will not work for your device. Because of its flash properties there is no support for storage.

You always need a scratch partition (swap using scratch) or a larger image-0 partition (swap using move) for mcuboot to work.

andy9a9, 2023-06-30:
uff, ok, I was thinking, that is compatible, but :(. So is there any way to use a OTA update including switching partitions for this device at all?

Laczen, 2023-06-30:
I think mcumgr (smp_svr) should work, but since there will be no storage possibility the usage of zephyr for this device will always be very limited.

andy9a9, 2023-06-30:
OK, but I made an another test, but it's still failing.
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* 256KB for bootloader. This is way to large but there is no way to make part of it usefull*/
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(256)>;
			read-only;
		};

		/* application image slot: 256KB */
		slot0_partition: partition@40000 {
			label = "image-0";
			reg = <0x00040000 DT_SIZE_K(256)>;
		};

		/* backup slot: 256KB */
		slot1_partition: partition@80000 {
			label = "image-1";
			reg = <0x00080000 DT_SIZE_K(256)>;
		};
	};
};
```
- mcuboot
```
west build -b nucleo_f756zg bootloader/mcuboot/boot/zephyr -d build-mcuboot
west flash -d build-mcubot

*** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
I: Starting bootloader
I: Primary image: magic=bad, swap_type=0x4, copy_done=0x2, image_ok=0x2
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
W: Failed reading image headers; Image=0
E: Unable to find bootable image
```
- hello_world
```
west build -b nucleo_f756zg zephyr/samples/hello_world -d build-hello-signed -- -DCONFIG_BOOTLOADER_MCUBOOT= -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem
west flash -d build-hello-signed

*** Booting Zephyr OS build v3.4.0-rc1-90-g5c7d2f1a90a6 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
E: Image in the primary slot is not valid!
E: Unable to find bootable image
```
Is it still a layout problem?

Laczen, 2023-06-30:
@andy9a9 The scratch partition is still missing, so you won't be able to do any updates.
The option `-DCONFIG_BOOTLOADER_MCUBOOT=` seems to be missing a "y"

andy9a9, 2023-06-30:
@Laczen Sorry, I copied it wrongly from CLI:
`west build -b nucleo_f756zg zephyr/samples/hello_world -d build-hello-signed -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\"`.

On the other hand, could you please post the correct flash layout?

Laczen, 2023-06-30:
@andy9a9, I have updated to proposed good pattition to include scratch instead of a second slot1.

andy9a9, 2023-06-30:
@Laczen After I used your layout, it works fine even if I removed my hack. So now I think, we can close the issue.
Next question is, if you wanted to "make the layout public".

Laczen, 2023-07-01:
> @Laczen Next question is, if you wanted to "make the layout public".

I don't see any problem with that, you can create a PR if you would like. I will leave it up to the st maintainers to approve.

erwango, 2023-07-03:
> > @Laczen Next question is, if you wanted to "make the layout public".
> 
> I don't see any problem with that, you can create a PR if you would like. I will leave it up to the st maintainers to approve.

Thanks @Laczen for the support.
@andy9a9 this would be welcome, thanks!

PR #59969 (2023-07-04): boards: arm: nucleo_f756zg: add flash0 partitions
Split `flash0` into partitions (4x256KB) to enable feature for switching between 2 applications.

Closes #59776
closed 2023-07-04 by fabiobaltieri