zephyrproject-rtos/zephyr #59788: `bt_unpair()` call in security changed callback causes assertion in `keys.c`

mkapala-nordic, 2023-06-27:
**Describe the bug**
Assert is triggered in the `smp_pairing_complete()` context after application removes the bonding information with the `bt_unpair()` API in the context of security changed callback.
The stack tries updating the keys usage database with the internal `bt_keys_store()` to finalize the Bluetooth pairing flow (`smp_pairing_complete()`). 
As the pairing keys are removed during the `bt_unpair()` API call, this operation results in the assert as the keys structure pointer is NULL.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify `samples/bluetooth/peripheral_hids` by commenting out `bt_conn_set_security()` and adding `bt_unpair()` in `security_changed()` callback:
```
diff --git a/samples/bluetooth/peripheral_hids/src/main.c b/samples/bluetooth/peripheral_hids/src/main.c
index 13109d40d9..2ecbd65a37 100644
--- a/samples/bluetooth/peripheral_hids/src/main.c
+++ b/samples/bluetooth/peripheral_hids/src/main.c
@@ -44,9 +44,9 @@ static void connected(struct bt_conn *conn, uint8_t err)
 
 	printk("Connected %s\n", addr);
 
-	if (bt_conn_set_security(conn, BT_SECURITY_L2)) {
-		printk("Failed to set security\n");
-	}
+	// if (bt_conn_set_security(conn, BT_SECURITY_L2)) {
+	// 	printk("Failed to set security\n");
+	// }
 }
 
 static void disconnected(struct bt_conn *conn, uint8_t reason)
@@ -71,6 +71,11 @@ static void security_changed(struct bt_conn *conn, bt_security_t level,
 		printk("Security failed: %s level %u err %d\n", addr, level,
 		       err);
 	}
+
+	printk("bt_unpair start\n");
+	int err2 = bt_unpair(BT_ID_DEFAULT, bt_conn_get_dst(conn));
+	printk("bt_unpair end, return: %d\n", err2);
+
 }
 
 BT_CONN_CB_DEFINE(conn_callbacks) = {

```

2. Build:
```
west build --pristine -d build -b nrf52840dk_nrf52840 -- -DCONFIG_BT_SMP_LOG_LEVEL_DBG=y -DCONFIG_BT_KEYS_LOG_LEVEL_DBG=y -DCONFIG_ASSERT=y -DCONFIG_SAMPLE_BT_USE_AUTHENTICATION=n -DCONFIG_LOG_PROCESS_THREAD_STACK_SIZE=8192
```

3. Flash:
```
west flash --skip-rebuild --erase -d build
```

4. Bond with the device (`Test HoG mouse`) using `nRF Connect for Mobile` or system level pairing.

**Expected behavior**
Host desists from keys usage database update using `bt_keys_store()` when there is nothing to be stored (`keys == NULL`).

**Logs:**
```
*** Booting Zephyr OS build zephyr-v3.4.0-433-g3be68bd4fe48 ***
[00:00:00.267,974] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.268,005] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.268,005] <inf> fs_nvs: data wra: 0, 0
[00:00:00.270,568] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.270,599] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.270,629] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.4 Build 99
[00:00:00.271,057] <dbg> bt_smp: bt_smp_init: LE SC enabled
[00:00:00.271,240] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.271,759] <inf> bt_hci_core: Identity: D1:C2:0E:ED:76:AB (random)
[00:00:00.271,789] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.271,820] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Advertising successfully started
[00:00:00.542,724] <dbg> bt_smp: bt_smp_pkey_ready: 
[00:00:14.026,031] <dbg> bt_keys: bt_keys_find_irk: 40:FC:FE:F3:A0:B8 (random)
[00:00:14.026,214] <dbg> bt_keys: bt_keys_find_irk: No IRK for 40:FC:FE:F3:A0:B8 (random)
[00:00:14.026,275] <dbg> bt_smp: bt_smp_accept: conn 0x20001238 handle 0
[00:00:14.026,275] <dbg> bt_smp: bt_smp_connected: chan 0x20001504 cid 0x0006
[00:00:14.026,519] <dbg> bt_keys: bt_keys_find_addr: 40:FC:FE:F3:A0:B8 (random)
Connected 40:FC:FE:F3:A0:B8 (random)
[00:00:14.233,337] <dbg> bt_smp: bt_smp_recv: Received SMP code 0x01 len 6
[00:00:14.233,367] <dbg> bt_smp: smp_pairing_req: req: io_capability 0x04, oob_flag 0x00, auth_req 0x2D, max_key_size 0x10, init_key_dist 0x0F, resp_key_dist 0x0F
[00:00:14.233,551] <dbg> bt_keys: bt_keys_get_addr: 40:FC:FE:F3:A0:B8 (random)
[00:00:14.233,764] <dbg> bt_keys: bt_keys_get_addr: created 0x20001c4c for 40:FC:FE:F3:A0:B8 (random)
[00:00:14.234,985] <dbg> bt_smp: smp_init: prnd ea45b8b30d1bdfcd7e769d65e70ad9d2
[00:00:14.235,015] <dbg> bt_smp: smp_pairing_req: rsp: io_capability 0x03, oob_flag 0x00, auth_req 0x09, max_key_size 0x10, init_key_dist 0x02, resp_key_dist 0x00
[00:00:14.332,214] <dbg> bt_smp: bt_smp_recv: Received SMP code 0x0c len 64
[00:00:14.332,244] <dbg> bt_smp: smp_public_key: 
[00:00:14.428,466] <dbg> bt_smp: bt_smp_recv: Received SMP code 0x04 len 16
[00:00:14.428,497] <dbg> bt_smp: smp_pairing_random: 
[00:00:14.592,376] <dbg> bt_smp: bt_smp_dhkey_ready: 0x20008414
[00:00:16.963,226] <dbg> bt_smp: bt_smp_recv: Received SMP code 0x0d len 16
[00:00:16.963,256] <dbg> bt_smp: smp_dhkey_check: 
[00:00:16.968,658] <dbg> bt_smp: sc_smp_send_dhkey_check: 
[00:00:17.694,580] <dbg> bt_keys: bt_keys_clear: 40:FC:FE:F3:A0:B8 (random) (keys 0x0000)
[00:00:17.695,037] <dbg> bt_keys: bt_keys_get_addr: 40:FC:FE:F3:A0:B8 (random)
[00:00:17.695,220] <dbg> bt_keys: bt_keys_get_addr: created 0x20001c4c for 40:FC:FE:F3:A0:B8 (random)
[00:00:17.695,251] <dbg> bt_smp: bt_smp_encrypt_change: chan 0x20001504 conn 0x20001238 handle 0 encrypt 0x01 hci status 0x00
Security changed: 40:FC:FE:F3:A0:B8 (random) level 2
bt_unpair start
[00:00:17.695,861] <dbg> bt_keys: bt_keys_clear: 40:FC:FE:F3:A0:B8 (random) (keys 0x0020)
bt_unpair returned: 0
[00:00:17.792,022] <dbg> bt_smp: smp_ident_info: 
--- 1 messages dropped ---
[00:00:17.792,205] <dbg> bt_keys: bt_keys_get_type: type 2 40:FC:FE:F3:A0:B8 (random)
[00:00:17.792,449] <dbg> bt_keys: bt_keys_find: type 2 40:FC:FE:F3:A0:B8 (random)
[00:00:17.792,633] <dbg> bt_keys: bt_keys_get_addr: 40:FC:FE:F3:A0:B8 (random)
[00:00:17.792,877] <dbg> bt_keys: bt_keys_get_addr: created 0x20001c4c for 40:FC:FE:F3:A0:B8 (random)
[00:00:17.792,999] <dbg> bt_smp: bt_smp_recv: Received SMP code 0x09 len 7
[00:00:17.793,212] <dbg> bt_smp: smp_ident_addr_info: identity 6C:F7:84:9A:F9:CB (public)
[00:00:17.793,487] <dbg> bt_keys: bt_keys_find_addr: 6C:F7:84:9A:F9:CB (public)
[00:00:17.793,670] <dbg> bt_keys: bt_keys_get_type: type 2 40:FC:FE:F3:A0:B8 (random)
[00:00:17.793,853] <dbg> bt_keys: bt_keys_find: type 2 40:FC:FE:F3:A0:B8 (random)
[00:00:17.794,036] <dbg> bt_keys: bt_keys_find_addr: 6C:F7:84:9A:F9:CB (public)
[00:00:17.794,860] <dbg> bt_smp: smp_pairing_complete: status 0x0
ASSERTION FAIL [keys != ((void *)0)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/keys.c:326
[00:00:17.794,921] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000146  r2/a3:  0x0000000e
[00:00:17.794,921] <err> os: r3/a4:  0x20000c58 r12/ip:  0x200069e0 r14/lr:  0x0001116d
[00:00:17.794,952] <err> os:  xpsr:  0x61000000
[00:00:17.794,952] <err> os: Faulting instruction address (r15/pc): 0x00027f7a
[00:00:17.795,013] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:17.795,043] <err> os: Current thread: 0x20001570 (unknown)
[00:00:17.969,390] <err> os: Halting system
```

**Environment:**
 - Ubuntu 22.04 LTS
 - Zephyr v3.4.0, commit: 3be68bd4fe48d4a3723ada18af9466bb646a5ad6
 - Tested on `nrf52840dk_nrf52840`
 
**Additional context**
It seems that BR counterpart (`smp_pairing_br_complete()`) does perform check before calling `bt_keys_store()`:
```
if (bond_flag && keys) {
	bt_keys_store(keys);
}
```
However, in `smp_pairing_complete()` there is no such check:
```
if (bond_flag) {
	bt_keys_store(conn->le.keys);
}
```

theob-pro, 2023-06-28:
The issue can only be reproduced when trying to bond with iOS devices, right @mkapala-nordic?

mkapala-nordic, 2023-06-28:
I reproduced it on both, iOS (iPhone XR, iOS 16.5) and Android (Redmi Note 10S, Android 11).

PR #62195 (2023-09-06): Bluetooth: Host: Fix SMP issues
Add a test to reproduce #59788 and #61465.

Fixes those two issues, if the bonding information has been cleared before pairing had a chance to complete (probably by the application), indicate this by setting an appropriate log message.

Also check that keys exist before calling `bt_keys_store`.

Fixes #59788 #61465
commit 8bd36fc5 (2023-09-06): Bluetooth: Host: Fixes SMP issues

If the bonding information has been cleared before pairing had a chance
to complete (probably by the application), indicate this by setting an
appropriate log message.

Also check that keys exist before calling `bt_keys_store`.

Fixes #59788 and #61465
closed 2023-06-29 by fabiobaltieri