zephyrproject-rtos/zephyr #59794: Merge hex file failed when building tf-m on nucleo_l552ze_q_ns

sjl3110, 2023-06-27:
Hi,

I referred with this guide: https://docs.zephyrproject.org/latest/boards/arm/nucleo_l552ze_q/doc/nucleol552ze_q.html

I have pulled the latest main repo, HEAD is `6050a10f8b298fe6994bb47ab9775ba535a604dd`. My build command on zephyr root path is
```
west build -b nucleo_l552ze_q_ns samples/tfm_integration/tfm_regression_test
```

The build is success but when it tries to merge to hex file, it failed.

The log is:

```
[557/650] Linking C executable bin/bl2.axf
Memory region         Used Size  Region Size  %age Used
           FLASH:       31812 B        54 KB     57.53%
     FLASH_NOHDP:         232 B         2 KB     11.33%
             RAM:       26960 B        63 KB     41.79%
[578/650] Linking C executable bin/tfm_s.axf
Memory region         Used Size  Region Size  %age Used
           FLASH:      174040 B       215 KB     79.05%
             RAM:       64788 B       127 KB     49.82%
[644/650] Linking C executable bin/tfm_ns.axf
Memory region         Used Size  Region Size  %age Used
           FLASH:       57596 B       159 KB     35.37%
             RAM:       21728 B       128 KB     16.58%
[648/650] Generating tfm_s_signed.bin
image.py: sign the payload
[649/650] Generating tfm_ns_signed.bin
image.py: sign the payload
[650/650] Generating tfm_s_ns_signed.bin
[15/168] Performing install step for 'tfm'
-- Install configuration: "MinSizeRel"
[163/168] Linking C executable zephyr/zephyr_pre0.elf

[168/168] Linking C executable zephyr/zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr.map zephyr/zephyr.hex zephyr/zephyr.bin zephyr/zephyr.lst zephyr/zephyr.stat zephyr/tfm_s_signed.hex zephyr/zephyr_ns_signed.hex zephyr/tfm_merged.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr.map /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr.bin /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr.lst /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr.stat /Users/sjl/zephyrproject/zephyr/build/zephyr/tfm_s_signed.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr_ns_signed.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/tfm_merged.hex 
: && ccache /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -L/usr/local/Homebrew/opt/libomp/lib   -gdwarf-4 zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj -o zephyr/zephyr.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  tfm/secure_fw/s_veneers.o  -fuse-ld=bfd  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr_final.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  modules/trusted-firmware-m/libtfm_api.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v8-m.main/nofp"  -L/Users/sjl/zephyrproject/zephyr/build/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  tfm/tf-m-tests/app/libtfm_api_ns.a  tfm/platform/ns/libplatform_ns.a  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /Users/sjl/zephyrproject/zephyr/build/zephyr && /usr/local/Homebrew/Cellar/cmake/3.24.3/bin/cmake -E copy zephyr_final.map zephyr.map && /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --gap-fill 0xff --output-target=ihex --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.hex && /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --gap-fill 0xff --output-target=binary --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.bin && /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -d -S zephyr.elf > zephyr.lst && /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf -e zephyr.elf > zephyr.stat && /Users/sjl/zephyrproject/zephyr/build/tfm/postbuild.sh /Users/sjl/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc && /usr/local/Homebrew/Cellar/cmake/3.24.3/bin/cmake -E env PYTHONPATH=/Users/sjl/zephyrproject/bootloader/mcuboot/scripts /usr/local/Homebrew/opt/python@3.10/bin/python3.10 /Users/sjl/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py --layout /Users/sjl/zephyrproject/zephyr/build/tfm/bl2/ext/mcuboot/CMakeFiles/signing_layout_ns.dir/signing_layout_ns.o -k /Users/sjl/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072_1.pem --public-key-format full --align 1 -v 0.0.0+0 --pad --pad-header --hex-addr=201326592 -s auto --measured-boot-record -H 0x400 /Users/sjl/zephyrproject/zephyr/build/tfm/bin/tfm_ns.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr_ns_signed.hex && /usr/local/Homebrew/Cellar/cmake/3.24.3/bin/cmake -E env PYTHONPATH=/Users/sjl/zephyrproject/bootloader/mcuboot/scripts /usr/local/Homebrew/opt/python@3.10/bin/python3.10 /Users/sjl/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py --layout /Users/sjl/zephyrproject/zephyr/build/tfm/bl2/ext/mcuboot/CMakeFiles/signing_layout_s.dir/signing_layout_s.o -k /Users/sjl/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072.pem --public-key-format full --align 1 -v 0.0.0+0 --pad --pad-header -s auto --measured-boot-record -H 0x400 /Users/sjl/zephyrproject/zephyr/build/tfm/bin/tfm_s.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/tfm_s_signed.hex && /usr/local/Homebrew/opt/python@3.10/bin/python3.10 /Users/sjl/zephyrproject/zephyr/scripts/build/mergehex.py -o /Users/sjl/zephyrproject/zephyr/build/zephyr/tfm_merged.hex /Users/sjl/zephyrproject/zephyr/build/tfm/bin/bl2.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/tfm_s_signed.hex /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr_ns_signed.hex
Memory region         Used Size  Region Size  %age Used
           FLASH:       18436 B       512 KB      3.52%
             RAM:        4184 B       192 KB      2.13%
        IDT_LIST:          0 GB         2 KB      0.00%
preprocess bl2 file
 updated
 updated
 updated
 updated
 updated
 updated
 updated
 updated
 updated
 updated
image.py: sign the payload
image.py: sign the payload
Traceback (most recent call last):
  File "/Users/sjl/zephyrproject/zephyr/scripts/build/mergehex.py", line 28, in merge_hex_files
    ih.merge(to_merge, overlap=overlap)
  File "/usr/local/Homebrew/lib/python3.10/site-packages/intelhex/__init__.py", line 875, in merge
    raise AddressOverlapError(
intelhex.AddressOverlapError: Data overlapped at address 0xC000000

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/sjl/zephyrproject/zephyr/scripts/build/mergehex.py", line 56, in <module>
    main()
  File "/Users/sjl/zephyrproject/zephyr/scripts/build/mergehex.py", line 52, in main
    merge_hex_files(args.output, args.input_files, args.overlap)
  File "/Users/sjl/zephyrproject/zephyr/scripts/build/mergehex.py", line 30, in merge_hex_files
    raise AddressOverlapError("{} has merge issues".format(hex_file_path))
intelhex.AddressOverlapError: /Users/sjl/zephyrproject/zephyr/build/zephyr/zephyr_ns_signed.hex has merge issues
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/Homebrew/bin/cmake --build /Users/sjl/zephyrproject/zephyr/build
```

FYI: The SDK version is 0.15.1

Thanks.

erwango, 2023-07-28:
Can you have a try with following overlay:

```
/ {
	chosen {
		zephyr,code-partition = &slot1_partition;
	};
};

&flash0 {

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(80)>;
			read-only;
		};
		/* Secure image primary slot */
		slot0_partition: partition@14000 {
			label = "image-0";
			reg = <0x00014000 DT_SIZE_K(224)>;
		};
		/* Non-secure image primary slot */
		slot1_partition: partition@4c000 {
			label = "image-1";
			reg = <0x0004c000 DT_SIZE_K(172)>;
		};
		/*
		 * The flash starting at 0x7F000 and ending at
		 * 0x80000 is reserved for the application.
		 */
		storage_partition: partition@77000 {
			label = "storage";
			reg = <0x0007F000 DT_SIZE_K(40)>;
		};
	};
};

``
I've picked up this layout from following config:
https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/bad0696f388d6f713742c5da8dfce426ec17ef20/platform/ext/target/stm/nucleo_l552ze_q/include/flash_layout.h#L38

sjl3110, 2023-08-01:
It works for me, great thanks : )

erwango, 2023-08-21:
Reopening as we need to fix this upstream

PR #63444 (2023-10-10): samples: tfm_regression_test: Provide a working partition for nucleo_l5
This partition is designed to work regression TFM configuration.

Fixes #59794
commit 1a006368 (2023-10-10): samples: tfm_regression_test: Provide a working partition for nucleo_l5

This partition is designed to work regression TFM configuration.

Fixes #59794
closed 2023-10-10 by carlescufi