zephyrproject-rtos/zephyr #59800: Driver for the LCD of the stm32f412g_disco board

supcik, 2023-06-27:
The [STM32F412G_DISCO](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/stm32f412g_disco/doc/index.rst) board is supported by Zephyr, but the LCD driver seems to be missing.

[image]

According to [the source file of the STM32Cube implementation of the BSP](https://github.com/STMicroelectronics/STM32CubeF4/blob/bb621289be5dd54f02e3198e1147b35aed4f7195/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery_lcd.c), the LCD driver can be either a [ST7789H2](https://github.com/STMicroelectronics/STM32CubeF4/blob/bb621289be5dd54f02e3198e1147b35aed4f7195/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery_lcd.c#L136) or a [LS016B8UY](https://github.com/STMicroelectronics/STM32CubeF4/blob/bb621289be5dd54f02e3198e1147b35aed4f7195/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery_lcd.c#L161), but it appears that all recent boards have the ST7789H2 controller.

On this board, the LCD controller is connected using [parallel interface in 8080/6800 modes](https://github.com/zephyrproject-rtos/zephyr/blob/abc5f4cad2eeba1a4992e9368247b8a72f732c0d/boards/arm/stm32f412g_disco/doc/index.rst?plain=1#L60)

Zephyr does have a [driver for the ST7789V](https://github.com/zephyrproject-rtos/zephyr/blob/abc5f4cad2eeba1a4992e9368247b8a72f732c0d/drivers/display/display_st7789v.c
) (hopefully compatible with the ST7789H2), but as far as I understand, this driver only supports SPI and not the required parallel interface.

The parallel interface for STM32 seems to be available in the [memc](https://github.com/zephyrproject-rtos/zephyr/blob/abc5f4cad2eeba1a4992e9368247b8a72f732c0d/drivers/memc/memc_stm32_sdram.c
) module of Zephyr, but I would have to investigate more to be sure.

I am ready to start implementing this driver myself (for my first contribution to Zephyr). With this issue, I would like to make sure that the maintainers are interested in this contribution and to check if other developers are already working on a similar driver.

supcik, 2023-08-30:
Please, remove the Stale label. I am still working on this driver.

stefbo, 2024-02-03:
Hi there, is there any progress on this or has development stopped?