zephyrproject-rtos/zephyr #59806: drivers: adc: enable blocking channel setup

maksimdrachov, 2023-06-28:
**Describe the bug**
I'm using the board `atsamc21n_xpro`, I was trying to read a pin voltage using ADC.

I was getting some strange results, so after some investigation, I have found that the issue was that the channel was not being set up correctly.

This is how I'm setting up my channel (please tell if I did something wrong, I stuck as close to the sample as possible):

```cpp
...
#define MY_ADC DT_NODELABEL(adc0)

static const struct adc_channel_cfg ch3_cfg_dt =
    ADC_CHANNEL_CFG_DT(DT_CHILD(DT_NODELABEL(adc0), channel_3));

static const struct adc_dt_spec adc_spec =
    ADC_DT_SPEC_GET_BY_IDX(DT_PATH(zephyr_user), 0);

int main()
{
    int err;
    const struct device* adc_dev = DEVICE_DT_GET(MY_ADC);

    /* Configure ADC channel prior to sampling */
    if (!device_is_ready(adc_dev))
    {
        printk("ADC device is not ready\n");
        return -ENODEV;
    }

    err = adc_channel_setup(adc_dev, &ch3_cfg_dt);
    if (err < 0) {
        printk("ADC channel setup error %d\n", err);
        return 0;
    }
...
```

**To Reproduce**

1. Let's say we want to set the reference voltage to `ADC_REF_EXTERNAL0`:

```
## .overlay file
/ {
	zephyr,user {
		/* EXT-1, pin 3 ADC(+) */
		io-channels = <&adc0 3>;
	};
};

&adc0 {
	#address-cells = <1>;
	#size-cells = <0>;

	channel@3 {
		reg = <3>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_EXTERNAL0";
		zephyr,vref-mv = <3300>;
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
		zephyr,input-positive = <3>;
	};
};
```

2. Set a breakpoint on `adc_sam0_channel_setup` in `zephyr/drivers/adc/adc_sam0.c`

Upon hitting this breakpoint, check if the ENABLE bit is set: 

[image]

If it is, then some of the settings won't be updated, see datasheet:

[image]

**Expected behavior**
The way I have solved the bug for now, is to make sure the ENABLE is disabled when starting the configuration:

[image]

**Impact**
Annoying bug

**Environment (please complete the following information):**
 - OS: MacOS/Linux
 - Commit SHA: `c22c4fd49f`

attie-argentum, 2023-07-05:
Hi @maksimdrachov, thanks for reporting this issue.

Please could you remove your fix, try adding `#define ADC_SAM0_REFERENCE_ENABLE_PROTECTED` to `soc/arm/atmel_sam0/samc21/soc.h` and report back? (see [`.../saml21/soc.h`](https://github.com/zephyrproject-rtos/zephyr/blob/77c86e8ad3e45644725df1da3a83bfffda516b83/soc/arm/atmel_sam0/saml21/soc.h#L43) for reference)

If that does the trick, then please feel free to open a PR, or I'd be happy to if you'd prefer.

attie-argentum, 2023-09-04:
PR in progress, see #60197

closed 2023-09-20 by maksimdrachov