zephyrproject-rtos/zephyr #59822: Memory partitioning on stm32h747i

TayyabaZainab0807, 2023-06-28:
I am new to the zephyr and west. I am trying to program my stm32h747i_disco device.
I use: west build -p always -b stm32h747i_disco_m7 zephyr/samples/basic/hello_world/
but it give me following partitioning:

           FLASH:      126480 B         1 MB     12.06%
             RAM:      434624 B       512 KB     82.90%
            QSPI:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:         16 KB        32 KB     50.00%
           SRAM4:          0 GB        64 KB      0.00%
          SDRAM2:          0 GB        32 MB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
        
Although stm32h747i_disco has 2MB flash and 1MB RAM, I don't want this partitioning between two core as I am only using one m7.
I am wondering what needs to be done to get 2 MB flash and  1MB ram on core m7?

FRASTM, 2023-06-30:
Hi @TayyabaZainab0807 

- For the RAM, you have access to the 864K. Did you try to add the node for partitionning the TCM RAM (dedicated to the cortex M7) ?

- For the Flash, did you try to redefine the flash

```
		flash-controller@52002000 {
			flash0: flash@8000000 {
				reg = <0x08000000 DT_SIZE_K(2048)>;
			};
		};
```

jhedberg, 2023-07-04:
@erwango @FRASTM seems like this should be converted into a discussion, since it seems more of a question rather than a bug?