zephyrproject-rtos/zephyr #59826: Cortex-M SMP support for Raspberry Pi Pico

carlescufi, 2023-06-28:
**Is your enhancement proposal related to a problem? Please describe.**

SMP in Cortex-M is not currently supported for the RPi Pico.

**Describe the solution you'd like**

Implement SMP specifically for Cortex-M on the RPi Pico.

**Additional context**

Requires that the aarch32 architecture transitions to `USE_SWITCH`

thedjnK, 2023-06-28:
There was a discussion on discord about this whereby it was mentioned that the pico SMP support was essentially bespoke to the pico because SMP support on ARM is not an official feature, are there any other *similar* ARM SMP cores where parts could be shared or would it be entirely for this CPU only?

danielallstar, 2024-04-09:
Hey all,

I am using the Raspberry Pi Pico in combination with Zephyr and want to make use of the 2nd core to run a function.
My initial thought was using SMP functionality within Zephyr to assign a working thread on both cores.
To my suprise however I found out that for the Raspberry Pi Pico only 1 core is defined.

My second thought if this functionality is not supported was to make use of the Raspberry Pi SDK, since
the SDK contains functionality to map a function call to the 2nd processing core.
The Pi Pico SDK HAL is available in the modules folder, but I can however not manage to include the right 
files.

Does anybody know how to make use of both processing cores within Zephyr?

Any help will be much apreciated!

@carlescufi @ithinuel @thedjnK @soburi

soburi, 2024-04-10:
Hi, @danielallstar 

> Does anybody know how to make use of both processing cores within Zephyr?

The simple answer is "not yet worked".

We use the PicoSDK driver part only.
The PicoSDK's boot code that including SMP setup are not used by zephyr.
We need to enhancing the boot code for supporting Cortex-M SMP , but not yet completed.

danielallstar, 2024-04-10:
@soburi, Thank you for your response. It is much appreciated!

If I understand correctly you do make use of the PicoSDK driver in Zephyr?

The PicoSDK hal files (HAL_PI_PICO) are loaded in the modules folder within Zephyr containing the needed multicore.h file to make use of the second core.
I do however not manage to include the multicore.h file from these files.
How did you manage to get the PicoSDK working within Zephyr?

Kind regards,

Daniël

ithinuel, 2024-04-10:
Hi @danielallstar ,

With regards to SMP support, the implementation is not as straight forward for Cortex-M0/+ as it may be for newer armv7m or armv8m.
Because Cortex-M0/+ lack the exclusive monitor implementation, the core cannot be soundly synchronised using a shared memory based spinlock as the Kernel currently relies on.
To enable SMP on the rp2040, some changes are required in the kernel to enable SoC specific synchronisation primitive (eg using the Spinlocks available in the rp2040's SIO block).

warasilapm, 2024-04-10:
@ithinuel I haven't been as locked in on this as I maybe want to have been - wasn't there a draft PR for this SoC kernel extension somewhere?

Rhvarrier, 2024-05-04:
> Hi @danielallstar ,
>
> With regards to SMP support, the implementation is not as straight forward for Cortex-M0/+ as it may be for newer armv7m or armv8m.
> Because Cortex-M0/+ lack the exclusive [...]

Hi

Isn't there support for AMP either. As this could allow to use the second even though its not as ideal?