zephyrproject-rtos/zephyr #59843: twister/sysbuild: `extra_config` gives malformed Kconfig if `sysbuild: true` in sample.yaml

nvlsianpu, 2023-06-29:
**bug description**
In the twister build extra_config paramiters gives malformed Kconfig while this is used along with the sysbuid.

For instance for fallowing sampe.yaml file
``` yaml
tests:
  sample.sysbuild_extra_config.faill
    sysbuild: true
    platform_allow: nrf5340dk_nrf5340_cpuapp
    integration_platforms:
      - nrf5340dk_nrf5340_cpuapp
    extra_configs:
      - CONFIG_CONSOLE=n
```

twister will fail on `warning: ignoring malformed line 'CONFIG_CONSOLE=n`

**To Reproduce**
Use extra_config in twister build along with sysbuild.

**Expected behavior**
extra_config should be transfer to the build similar as for simple builds.

**Impact**
annoyance.
Looks like we can use `extra_args` instead, but this is apparently a bug.

**Environment (please complete the following information):**

savvn001, 2023-07-19:
Having same issue. Adding anything under `extra_configs` will cause this `warning: ignoring malformed line` problem

nordicjm, 2023-08-03:
Cannot replicate. I added:
```
    extra_configs:
      - CONFIG_CONSOLE=n
```
To https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/boot/mcuboot_recovery_retention and ran:
```
./scripts/twister -T tests/boot/mcuboot_recovery_retention/
ZEPHYR_BASE unset, using "/tmp/aa/zephyr"
Renaming output directory to /tmp/aa/zephyr/twister-out.94
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-1904-g6dc3a65e9b3e
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /tmp/aa/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    2/   2  100%  skipped:    0, failed:    0, error:    0
INFO    - 2 test scenarios (2 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 2 of 2 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 53.67 seconds
INFO    - In total 0 test cases were executed, 2 skipped on 1 out of total 594 platforms (0.17%)
INFO    - 0 test configurations executed on platforms, 2 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /tmp/aa/zephyr/twister-out/twister.json
INFO    - Writing xunit report /tmp/aa/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /tmp/aa/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

PerMac, 2023-09-06:
In fact, there are already tests using sysbuild and extra_configs and they do work. E.g. https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/boot/test_mcuboot/testcase.yaml#L27

nordicjm, 2023-09-06:
Closing as unable to reproduce, open if the issue becomes apparent in future.

closed 2023-09-06 by nordicjm