zephyrproject-rtos/zephyr #59848: Zbus: channel notify operation may block unexpectedly

jomigo96, 2023-06-29:
**Describe the bug**

Some Zbus operations, namely `zbus_chan_notify()`, may pend the calling thread even when used with `K_NO_WAIT`.

This is verified when there are subscribers enabled on a channel, but for whatever reason they are not consuming messages. Eventually the internal `queue` will be full and the thread attempting to notify will be left pending on this queue.

The following code sample demonstrates the circumstances:

```c
#include <zephyr/kernel.h>
#include <zephyr/zbus/zbus.h>

struct my_data {
	int data;
};

ZBUS_SUBSCRIBER_DEFINE(my_sub, 1);
ZBUS_CHAN_DEFINE(my_chan, struct my_data, NULL, NULL, ZBUS_OBSERVERS(my_sub), {0});

void timer_expired(struct k_timer *timer)
{
	printk("Whoops!\n");
}

K_TIMER_DEFINE(watchdog, timer_expired, NULL);

void main(void)
{
	while(1) {
		k_timer_start(&watchdog, K_MSEC(500), K_NO_WAIT);
		zbus_chan_notify(&my_chan, K_NO_WAIT);
	}
}
```

The timer is constantly reset and would not be expected to expire, but quickly the call to `zbus_chan_notify` blocks and this happens.
This can be reproduced with qemu (tested on qemu_x86_64, qemu_cortex_a53), but was originally detected on a STM32H743 board.

**To Reproduce**

Compile and run the code sample above.

```
west build -b qemu_x86_64 ... -DCONFIG_ZBUS=y -t run
```

**Expected behavior**

API functions that accept a timeout argument should not block when called with `K_NO_WAIT`. In the above sample, there should be no output.

**Impact**

Mild annoyance, can be bypassed by providing a short timeout.

**Possible causes**

Looks to be caused by a wrap-around on unsigned integer subtraction:

`subsys/zbus/zbus.c L12`
```c
k_timeout_t _zbus_timeout_remainder(uint64_t end_ticks)
{
	int64_t now_ticks = sys_clock_tick_get();

	return K_TICKS((k_ticks_t)MAX(end_ticks - now_ticks, 0));
}
```

The result of `end_ticks - now_ticks` is unsigned according to integer promotion rules, so the `MAX` is ineffective, and this results in a very large timeout.
A possible fix is to cast `end_ticks` to `int64_t`.

**Environment:**
 - OS: Ubuntu 22.04.2 LTS under WSL2
 - Toolchain: Zephyr 3.4 with SDK 0.16.1

jomigo96, 2023-06-29:
@rodrigopex

rodrigopex, 2023-07-05:
@jomigo96, please try to run using the code proposed on #60048 to check if it runs properly on your setup. Thanks.

jomigo96, 2023-07-06:
It does solve the issue, thanks!

rodrigopex, 2023-07-06:
> It does solve the issue, thanks!
> 

The PR is on the way. Thanks for contributing.

closed 2023-09-18 by rodrigopex