zephyrproject-rtos/zephyr #59850: net: offload: connect() on UDP socket with offloaded IP results in crash

mniestroj, 2023-06-29:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
Calling `connect()` on UDP socket with offloaded IP (e.g. with ESP-AT driver) results in crash. This is related to calling `net_context_recv()` before `net_context_connect()`, which was introduced as part of commit 2c7507036050804f207c0604073f38798bc1a5fd:
```
[00:00:02.124,328] <err> os: ***** MPU FAULT *****
[00:00:02.124,359] <err> os:   Instruction Access Violation
[00:00:02.124,359] <err> os: r0/a1:  0x20002be8  r1/a2:  0x0002225f  r2/a3:  0x00000000
[00:00:02.124,389] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000007 r14/lr:  0x0000c51f
[00:00:02.124,420] <err> os:  xpsr:  0x60000000
[00:00:02.124,420] <err> os: Faulting instruction address (r15/pc): 0xaaaaaaaa
[00:00:02.124,481] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:02.124,511] <err> os: Current thread: 0x20001318 (abc)
[00:00:03.020,660] <err> os: Halting system
```

Crash happens because `context->iface` is `-1` when `net_context_recv()` is called, so code fragment like (beginning of `net_context_recv()`):
```
	if (IS_ENABLED(CONFIG_NET_OFFLOAD) &&
	    net_if_is_ip_offloaded(net_context_get_iface(context))) {
		ret = net_offload_recv(
			net_context_get_iface(context),
			context, cb, timeout, user_data);
		goto unlock;
	}
```
results in undefined behavior, since `net_if_is_ip_offloaded()` is called on `NULL`.

Note that this is a regression caused by 2c7507036050804f207c0604073f38798bc1a5fd.

A workaround to make `connect()` work (not crash) is achieved by adding:
```
	k_mutex_lock(&context->lock, K_FOREVER);

+	ret = bind_default(context);
+	if (ret) {
+		goto unlock;
+	}
+
	if (IS_ENABLED(CONFIG_NET_OFFLOAD) &&
	    net_if_is_ip_offloaded(net_context_get_iface(context))) {
		ret = net_offload_recv(
			net_context_get_iface(context),
			context, cb, timeout, user_data);
		goto unlock;
	}
```
in `net_context_recv()`. This was done just to confirm that the issue is with socket not being bound to any specific interface at the time of calling `net_context_recv()` and this is (most probably) not a feasible fix to apply.

**To Reproduce**
Steps to reproduce the behavior:
1. `sock = zsock_socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);`
2. `ret = zsock_connect(sock, addr, addrlen);`
3. See crash

**Expected behavior**
`connect()` should succeed, since for UDP sockets all it should do is assigning the target address and network interface to be used.

**Impact**
Showstopper on using UDP sockets with `connect()` with ESP-AT WiFi driver.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.0
 - Commit SHA or Version used: v3.4.0 and tip of main (e484ab2fa8cbff320c907e21f079cf70c1d1ec29)

rlubos, 2023-06-29:
To me it seems that for offloaded case the interface on the context could (or even should) be set right after the "offloaded" context is allocated - `net_offload_get()` already takes an interface as an argument.

mniestroj, 2023-06-29:
> To me it seems that for offloaded case the interface on the context could (or even should) be set right after the "offloaded" context is allocated - net_offload_get() already takes an interface as an argument.

Makes sense! The assumption about using default offloaded interface is something to be fixed, but until that happens it looks reasonable to already call `net_context_set_iface()` in `net_context_get()` already. I've created #59851 which implements your suggestion!

rlubos, 2023-06-29:
> The assumption about using default offloaded interface is something to be fixed

Indeed, it won't play well with multi-interface systems.

PR #59851 (2023-06-29): net: context: set default offloaded iface during net_context_get()
Set default offloaded interface during `net_context_get()` call, so that
`net_context_recv()` can be called before `net_context_connect()`. There is
already an assumption about using default network interface, so this should
not be harmful.

Fixes: 2c7507036050 ("net: sockets: tcp: Fix possible race between
  connect/recv")
Fixes: #59850
PR #59857 (2023-07-19): [Backport v3.4-branch] net: context: set default offloaded iface during net_context_get()
Backport 40d224022608a3e5d68e95cc4d6f7606f1d48fb2 from #59851.

Fixes: #59850
closed 2023-06-29 by carlescufi