zephyrproject-rtos/zephyr #59865: shell cmd: "devmem dump" only works for the 1st call

botbotty, 2023-06-30:
**Describe the bug**
Shell command "devmem dump" works properly when I call it for the first time. But when I call the exact same command again, it shows an error `'-a <address>' is mandatory`.
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. cd "$(west topdir)/zephyr"
1. west build --board qemu_x86 --target run samples/hello_world/ -- -DCONFIG_SHELL=y
2. call command "devmem dump -a 0 -s 0x100" in the shell
3. above command works fine
4. call command "devmem dump -a 0 -s 0x100" in the shell again
5. above command doesn't work

**Expected behavior**
both calls to the command should work

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper): N/A

**Logs and console output**
```
-- west build: running target run
[3/12] Linking C executable zephyr/zephyr_pre0.elf

[11/12] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             RAM:       90144 B         3 MB      2.87%
        IDT_LIST:          0 GB         2 KB      0.00%
[11/12] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build zephyr-v3.4.0-554-g33b116407b03 ***
Hello World! qemu_x86

uart:~$ devmem dump -a 0 -s 0x100
Mapped 0x0 to 0x7fa000

007FA000: 53 ff 00 f0 53 ff 00 f0  c3 e2 00 f0 53 ff 00 f0 |S...S... ....S...|
007FA010: 53 ff 00 f0 54 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...T... S...S...|
007FA020: a5 fe 00 f0 87 e9 00 f0  33 d4 00 f0 33 d4 00 f0 |........ 3...3...|
007FA030: 33 d4 00 f0 33 d4 00 f0  57 ef 00 f0 33 d4 00 f0 |3...3... W...3...|
007FA040: 90 d2 00 f0 4d f8 00 f0  41 f8 00 f0 fe e3 00 f0 |....M... A.......|
007FA050: 39 e7 00 f0 59 f8 00 f0  2e e8 00 f0 d2 ef 00 f0 |9...Y... ........|
007FA060: 58 d4 00 f0 f2 e6 00 f0  6e fe 00 f0 53 ff 00 f0 |X....... n...S...|
007FA070: 53 ff 00 f0 53 ff 00 f0  bc 61 00 f0 a0 94 00 c0 |S...S... .a......|
007FA080: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA090: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0A0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0B0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0C0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0D0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0E0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
007FA0F0: 53 ff 00 f0 53 ff 00 f0  53 ff 00 f0 53 ff 00 f0 |S...S... S...S...|
uart:~$ devmem dump -a 0 -s 0x100
'-a <address>' is mandatory
uart:~$ devmem dump -a 0 -s 0x100
'-a <address>' is mandatory
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 33b116407b03883efd889f75d69f5bc753b7ade3

cfriedt, 2023-07-04:
This is likely a result of this https://github.com/zephyrproject-rtos/zephyr/issues/52700

cfriedt, 2023-11-03:
Still a problem, as getopt globals need to be in TLS or protected via pthread_key_create.

I would say, if we are to use pthread_key_create then we need to (mandatorily) unify pthreads and k_threads.

That actually wouldn't be all that bad.

cfriedt, 2024-01-03:
I need to check if this is still an issue. From what I recall, I was able to run it multiple times recently. There may have been improvements to the underlying `getopt()` function.

valentinkorenblit, 2024-01-16:
Hello, I was testing devmem dump and saw this issue but it works several times when enabling SHELL_GETOPT

cfriedt, 2024-01-16:
> Hello, I was testing devmem dump and saw this issue but it works several times when enabling SHELL_GETOPT

Thanks for confirming. That reflects my recent observations too.

I'll close this as complete.

closed 2024-01-16 by cfriedt