zephyrproject-rtos/zephyr #59881: wifi: esp32: build error due to `WIFI_SCAN_TYPE_{ACTIVE,PASSIVE}` redeclaration

mniestroj, 2023-06-30:
**Describe the bug**
Both Zephyr:
```
enum wifi_scan_type {
	WIFI_SCAN_TYPE_ACTIVE = 0,
	WIFI_SCAN_TYPE_PASSIVE,
};
```
and hal_espressif:
```
typedef enum {
    WIFI_SCAN_TYPE_ACTIVE = 0,  /**< active scan */
    WIFI_SCAN_TYPE_PASSIVE,     /**< passive scan */
} wifi_scan_type_t;
```
repositories declare the same enumerators. This results in build failure:
```
In file included from /modules/hal/espressif/zephyr/esp32/../esp_shared/include/wifi/esp_networking_priv.h:9,
                 from /zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:20:
/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/include/esp_wifi_types.h:108:5: error: redeclaration of enumerator 'WIFI_SCAN_TYPE_ACTIVE'
  108 |     WIFI_SCAN_TYPE_ACTIVE = 0,  /**< active scan */
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from /zephyr/include/zephyr/net/wifi_mgmt.h:16,
                 from /zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:17:
/zephyr/include/zephyr/net/wifi.h:268:9: note: previous definition of 'WIFI_SCAN_TYPE_ACTIVE' with type 'enum wifi_scan_type'
  268 |         WIFI_SCAN_TYPE_ACTIVE = 0,
      |         ^~~~~~~~~~~~~~~~~~~~~
/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/include/esp_wifi_types.h:109:5: error: redeclaration of enumerator 'WIFI_SCAN_TYPE_PASSIVE'
  109 |     WIFI_SCAN_TYPE_PASSIVE,     /**< passive scan */
      |     ^~~~~~~~~~~~~~~~~~~~~~
/zephyr/include/zephyr/net/wifi.h:269:9: note: previous definition of 'WIFI_SCAN_TYPE_PASSIVE' with type 'enum wifi_scan_type'
  269 |         WIFI_SCAN_TYPE_PASSIVE,
      |         ^~~~~~~~~~~~~~~~~~~~~~
/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c: In function 'esp32_wifi_scan':
/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:409:28: error: invalid type argument of '->' (have 'wifi_scan_config_t')
  409 |                 scan_config->scan_type = params->scan_type;
      |                            ^~
```

This is a regression introduced by 69fead199828a418c31a2f4f8362e6d94dc703ef which was part of #59204.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b esp32 samples/net/wifi`
2. See error

**Expected behavior**
WiFi driver for ESP32 platform should build without issue.

**Impact**
Showstopper for using WiFi on ESP32 platform with Zephyr.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0
 - Commit SHA: 33b116407b03883efd889f75d69f5bc753b7ade3

rlubos, 2023-07-04:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/59995

mniestroj, 2023-07-04:
> Fixed by #59995

No, it is not fixed. Referenced issue is fixing something else (though related).

rlubos, 2023-07-04:
Apologies, you're right.

It seems that @marekmatej has already handled the name collision on the ESP side (https://github.com/zephyrproject-rtos/hal_espressif/commit/d5a8fc861136abe74f05f56fad5738477a9985d2) so we'll need to wait for the `hal_espressiff` revision update in Zephyr.

marekmatej, 2023-07-04:
The issue was fixed on hal_espressif and already merged. Unfortunately, the fix commit sits on the top of changes that need https://github.com/zephyrproject-rtos/zephyr/pull/58454 in order to work.

PR #60633 (2023-07-25): west.yml: update hal_espressif to fix Wi-Fi label redeclaration
Fix Wi-Fi build due to label declaration in hal_espressif.

Fixes #59881
commit 0439dfa9 (2023-07-25): west.yml: update hal_espressif to fix Wi-Fi label redeclaration

Fix Wi-Fi build due to label declaration in hal_espressif.

Fixes #59881
closed 2023-07-25 by carlescufi