zephyrproject-rtos/zephyr #59882: NXP LPUART receiver disabled when disabling Rx interrupts

manuargue, 2023-06-30:
**Describe the bug**
When running `tests/drivers/uart/uart_basic_api.drivers.uart.basic_api` for the `mr_canhubk3` board being introduced in https://github.com/zephyrproject-rtos/zephyr/pull/58332, the testcase `test_uart_poll_in` cannot receive characters from the serial console.

**To Reproduce**
`west build -p auto -b mr_canhubk3 ./tests/drivers/uart/uart_basic_api -t flash`

**Expected behavior**
`test_uart_poll_in` can receive characters from the serial console.

**Impact**
When using `CONFIG_UART_INTERRUPT_DRIVEN`, UART poll cannot be used together with UART FIFO.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-567-gc3a9a886e1dc ***
Running TESTSUITE uart_basic_api
===================================================================
START - test_uart_config_get
This is a configure_get test.
 PASS - test_uart_config_get in 0.004 seconds
===================================================================
START - test_uart_configure
 PASS - test_uart_configure in 0.002 seconds
===================================================================
START - test_uart_fifo_fill
This is a FIFO test.
 PASS - test_uart_fifo_fill in 0.501 seconds
===================================================================
START - test_uart_fifo_read
Please send characters to serial console
 PASS - test_uart_fifo_read in 4.254 seconds
===================================================================
START - test_uart_poll_in
```
(UART cannot receive characters after this line)

**Environment (please complete the following information):**
 - OS: Linux, Windows
 - Toolchain Zephyr SDK 0.16.0
 - Commit SHA (rebased PR on) ad6fdaf2c23f9fc7206036e565db71ea0b2331d3 

**Additional context**
This seems to be a regression introduced in https://github.com/zephyrproject-rtos/zephyr/pull/50657/files#diff-b52eb85e1cdf037f16de983f18bf4cfb8079bc5e657706e94a36de8cf842167aR314 
In the test `tests/drivers/uart/uart_basic_api.drivers.uart.basic_api`:
1. `test_uart_fifo_read` runs first, and disables UART rx irq (also disabling the receiver) https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/drivers/uart/uart_basic_api/src/test_uart_fifo.c#L124 
2. `test_uart_poll_in` runs after, and cannot receive because receiver is disabled.
Reverting https://github.com/zephyrproject-rtos/zephyr/pull/50657/files#diff-b52eb85e1cdf037f16de983f18bf4cfb8079bc5e657706e94a36de8cf842167aR314 fixes this issue. I've checked all other UART drivers and none of them is changing the state of the receiver when enabling/disabling interrupts.

decsny, 2024-04-22:
raised priority because this actually breaks a lot of platforms

decsny, 2024-04-22:
It is actually truly amazing that this bug has existed for 2 years, I think it was never caught by automated testing because it requires user input so it was excluded, so I'll fix that too

PR #71900 (2024-04-30): drivers: uart_mcux_lpuart: Fix issues with different UART APIs breaking each other
This reverts commit https://github.com/zephyrproject-rtos/zephyr/commit/a80fa5163bb162f12614dc792aa17016c0fa1ac6
```
That commit broke the polling API when used with the interrupt driven,
it is completely wrong to disable the receiver from irq_rx_disable,
the uart_basic_api test has been broken on all lpuart platforms because
of this bug.
```

Fixes #59882

In addition, fix the issue that the original commit was trying to fix, by:
```
Handle controller instances wanting to use different
API instances, need to ch [... cut]
closed 2024-04-30 by aescolar