zephyrproject-rtos/zephyr #59921: Enableing PXP accelearion for lvgl for i. MXRT1064

Gigate, 2023-07-03:
Greetings all,
Currently it isn't possible to use pxp gpu acceleration for 2D lvgl rendering in ZephyrOS. Pxp acceleration is is not only necessary for faster rendering, but also for display rotation, which doesn't seem to be possible (with acceptable performance) without the pxp accel.

Here is some research of files that would be need. But just adding them to a Project isn't enough: 
- lv_gpu/lv_gpu_nxp.c, lv_gpu/lv_gpu_nxp.h: low level drawing calls for LVGL
- lv_gpu/lv_gpu_nxp_osa.c, lv_gpu/lv_gpu_osa.h: default implementation of OS-specific functions (bare metal and FreeRTOS only)
     - optional, required only if `LV_USE_GPU_NXP_PXP_AUTO_INIT` is set to 1
- PXP related code depends on two drivers provided by MCU SDK. These drivers need to be added to project:
    - fsl_pxp.c, fsl_pxp.h: PXP driver
    - fsl_cache.c, fsl_cache.h: CPU cache handling functions

arfoll, 2023-07-03:
@danieldegrasse this is what we discussed in Prague at EOSS :) If we can help or you can point us how to contribute here we'd be open to this.

danieldegrasse, 2023-07-03:
Hi @arfoll, I think integrating the PXP into LVGL directly would come later, as there are additional functions the PXP supports (like blending) that LVGL leverages. For the display rotation use case, I will look into exposing the PXP driver using the DMA API to see if it can satisfy your needs there.

PR #60064 (2023-07-25): drivers: dma: add DMA driver for NXP PXP engine
The NXP Pixel Pipeline Engine (PXP) is a 2D DMA engine capable of
accelerating display rotation, color space conversion, and limited
2D blending operations. This DMA driver only supports rotation of a
framebuffer, via a set of custom dma_slot values.

The driver enables rotation in concert with the ELCDIF driver. For example, a user can configure the ELCDIF driver to leverage the PXP to rotate the display by 270 degrees with the following settings:

```
CONFIG_DMA=y
CONFIG_MCUX_ELCDIF_PXP=y
CONFIG_MCUX_ELCDIF_PXP_ROTATE_270=y
```

If used, LVGL c [... cut]
commit d2372139 (2023-07-25): drivers: display: display_mcux_elcdif: enable display rotation using PXP

Enable display rotation using the NXP pixel pipeline (PXP). The ELCDIF
will only utilize the PXP if a framebuffer equivalent in size to the
screen is provided to display_write. The rotation angle can be
configured via Kconfig at build time.

Fixes #59921
closed 2023-07-25 by carlescufi