zephyrproject-rtos/zephyr #59922: kernel: spinlocks: clean up struct k_spinlock

cfriedt, 2023-07-03:
**Is your enhancement proposal related to a problem? Please describe.**
The existing workaround for `struct k_spinlock` occasionally being 0 bytes does not always work around things. In particular when you do pointer arithmetic, which is semi important in C, but triggers some warnings now that are showing up in gcc (not just g++).

Is there a simple change that can be made so this just works better?

Maybe some compiler experts know if it can be considered a size / runtime optimization? 🤷 

**Describe the solution you'd like**
Is there is another solution that works with like array indexing or one that does not require others to modify their source to make workarounds?

Perhaps there are some Macrobatics.

**Describe alternatives you've considered**
Patching API consumers whenever this issue is hit in the future.

**Additional context**
[image]

itagagaki, 2023-12-15:
@cfriedt What do you find unsatisfactory about the workaround by the char dummy?

cfriedt, 2023-12-15:
> @cfriedt What do you find unsatisfactory about the char dummy workaround?

It's in the description / screenshots / linked PR

itagagaki, 2023-12-15:
Sorry, I asked because I couldn't figure it out after reading through the whole thing. I want to know specifically what problem  remains after the workaround PR is merged.

cfriedt, 2023-12-19:
@itagagaki - the issue in this ticket is that pointer arithmetic does not work on spinlocks if they are zero bytes. So an array of spinlocks will not be addressable.

Another issue is that spinlocks are quite large these days, in particular with ticket spinlocks and spinlock validation.

Ideally, we could use only the bits required. However, Zephyr's atomics API only operates on longs.

The stdatomic.h API use C11 generics to support atomic operations on single bits, bytes, shorts, ints, and longs. So there is a significant opportunity to improve space-wise.

itagagaki, 2023-12-25:
I see. Thanks for the explanation.

The first problem, that if the size of struct k_spinlock is zero, data of that type is not be addressable, you solved this with your patch. To prevent this from happening, you made it so that it has 1 byte of dummy data if there is no other data. This data is useless, but there's no better way, right?

And another problem is that Zephyr's atomic API only implements long, right? If char is also supported, for example, the size of this structure can be shrunk, right?
If so, I think it would be better to have a separate topic to extend atomic, don't you?

cfriedt, 2023-12-30:
> The first problem, that if the size of struct k_spinlock is zero, data of that type is not be addressable, you solved this with your patch. To prevent this from happening, you made it so that it has [...]

I think there probably are better ways. E.g. using a union. Possibly a Kconfig option. Maybe `K_SPINLOCK_ARRAY_DEFINE()` or something along those lines.

It would be kind of good for `k_spinlock` to be addressable in general, otherwise it's getting into undefined behaviour. I do realize that there are maybe speed benefits to having zero size, but also being a core part of the kernel, it's important to be safe for safety critical systems.

Would be good to get a quantitative idea of the performance implications of making the size minimally 1 byte.

> And another problem is that Zephyr's atomic API only implements long, right?

There is a separate issue for that already (#39641)