zephyrproject-rtos/zephyr #59942: posix: implement `pthread_equal()`

ycsin, 2023-07-03:
There are currently 20 missing POSIX functions with trivial implementations, `pthread_equal()` is one of them.

> pthread_equal - compare thread IDs
> 
> This function shall compare the thread IDs t1 and t2.

See https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_equal.html

See RFC #51211 for more info.

jaiiarora, 2023-07-15:
I would like to work on this issue. Can you assign it to me?

cfriedt, 2023-07-15:
> I would like to work on this issue. Can you assign it to me?

@jaiiarora - we can only assign issues to contributors, but you are more than welcome to create a pull request 😁

Please remember to follow the [contributor guidelines (https://docs.zephyrproject.org/latest/contribute/index.html).

jaiiarora, 2023-07-16:
I have raised a PR @cfriedt 
If any update is needed, please let me know. I will work on it.

closed 2023-07-20 by carlescufi