zephyrproject-rtos/zephyr #59943: posix: implement `pthread_kill()`

ycsin, 2023-07-03:
There are currently 20 missing POSIX functions with trivial implementations, `pthread_kill()` is one of them.

> pthread_kill - send a signal to a thread
> 
> The pthread_kill() function shall request that a signal be delivered to the specified thread.
> 
> As in [kill()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/kill.html), if sig is zero, error checking shall be performed but no signal shall actually be sent.

See https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_kill.html

See RFC #51211 for more info.

AbhinavMir, 2024-01-18:
I'm thinking of doing this in a generic POSIX way, lmk if this sounds right: pthread_t can be converted into an index, with which we find the corresponding k_thread in the posix_thread_pool, and then send a signal to that thread. I'm just wondering how can I "send" a signal to the thread. `SIGTERM` or `SIGKILL` are the usual signals, but is there a signal handler I can interact with? @ycsin

cfriedt, 2024-01-18:
@AbhinavMir - I have a draft Pr for this already. It might make sense to attempt a different enhancement issue.

AbhinavMir, 2024-01-18:
Makes sense, thank you