zephyrproject-rtos/zephyr #59950: posix: implement `mmap()`

ycsin, 2023-07-03:
There are currently 20 missing POSIX functions with trivial implementations, `mmap()` is one of them.

> mmap - map pages of memory
> 
> The mmap() function shall establish a mapping between an address space of a process and a memory object.
> ...

See https://pubs.opengroup.org/onlinepubs/9699919799/functions/mmap.html

See RFC #51211 for more info.

KaiserS0ze, 2023-11-20:
Hi @ycsin I am a first-time contributor. Can I take a crack at this?

ycsin, 2023-11-20:
> Hi @ycsin I am a first-time contributor. Can I take a crack at this?

Hi, please feel free to submit PR for anything

KaiserS0ze, 2023-11-21:
Thanks. Yes, I think there are a lot of edge cases. I will start off with a simple implementation.

closed 2024-06-14 by nashif