zephyrproject-rtos/zephyr #59952: posix: implement `shm_open()`

ycsin, 2023-07-03:
There are currently 20 missing POSIX functions with trivial implementations, `shm_open()` is one of them.

> shm_open - open a shared memory object (REALTIME)
> 
> The shm_open() function shall establish a connection between a shared memory object and a file descriptor. It shall create an open file description that refers to the shared memory object and a file descriptor that refers to that open file description. The file descriptor shall be allocated as described in [File Descriptor Allocation](https://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_14), and can be used by other functions to refer to that shared memory object. The name argument points to a string naming a shared memory object. It is unspecified whether the name appears in the file system and is visible to other functions that take pathnames as arguments. The name argument conforms to the construction rules for a pathname, except that the interpretation of <slash> characters other than the leading <slash> character in name is implementation-defined, and that the length limits for the name argument are implementation-defined and need not be the same as the pathname limits {PATH_MAX} and {NAME_MAX}. If name begins with the <slash> character, then processes calling shm_open() with the same value of name refer to the same shared memory object, as long as that name has not been removed. If name does not begin with the <slash> character, the effect is implementation-defined.
> ...

See https://pubs.opengroup.org/onlinepubs/9699919799/functions/shm_open.html

See RFC #51211 for more info.

NeimadTL, 2023-09-20:
@cfriedt, @nashif , I'm new to embedded development but I'd like to take a stab at it. Is that simple enough for somebody like ? Could one of you walk me through this please ?

cfriedt, 2024-01-20:
> Is that simple enough for somebody with very little knowledge ? Could one of you walk me through this please ?

@NeimadTL - I can give you some pointers.

First, it's a great idea to join the Zephyr Discord to get real-time feedback from everyone in the `#posix` channel.

https://chat.zephyrproject.org/

If you are new to Zephyr, the [Getting Started Guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) is always the best place to start.

Otherwise, I would recommend looking at the https://opengroup.org page linked.

From there you could start with a testsuite and do a bit of test driven development.

Look at the function prototype. Add tests that cover the range of invalid and valid input, like a truth table. So if there are 3 arguments to a function, the truth table should have 2^3=8 cases. Many of those will be "degenerate" cases, but some may represent undefined behaviour. It's a good idea to explicitly call those out with a comment.

With valid input, there may still be many more interesting testpoints, which can be added with different `ZTEST()` test cases.

That being said, we usually like to see 2-3 commits in a PR

1. Implementation (in e.g. lib/ include/, Konfig, CMakeLists.txt)
2. Tests (under e.g. tests/posix/)
3. Doc updates (if needed, to mark a feature as supported)

There are a lot of ways to get hung in CI, so it's best to ensure the testsuite passes prior to making a PR (run tests with `west` and `twister`). 

Other than that, there are coding guidelines and some style conventions; use `scripts/checkpatch.pl` before submitting a PR.

Lots of info at https://docs.zephyrproject.org

In this particular case, you should add a "shm driver" for the `lib/is/fdtable.c` The API for that is in `include/zephyr/fdtable.h`. A good reference about how you can do that is in `lib/posix/eventfd.c`.

Additionally, for shared memory, you will probably want to add a new testsuite in `tests/posix/common/src/shm.c`.

Jump on discord if you have any questions and thanks for your help 👍

closed 2024-06-14 by nashif