zephyrproject-rtos/zephyr #59954: posix: implement `clock_getcpuclockid()`

ycsin, 2023-07-03:
There are currently 20 missing POSIX functions with trivial implementations, `clock_getcpuclockid()` is one of them.

> clock_getcpuclockid - access a process CPU-time clock (ADVANCED REALTIME)
> 
> The clock_getcpuclockid() function shall return the clock ID of the CPU-time clock of the process specified by pid. If the process described by pid exists and the calling process has permission, the clock ID of this clock shall be returned in clock_id.
> 
> If pid is zero, the clock_getcpuclockid() function shall return the clock ID of the CPU-time clock of the process making the call, in clock_id.
> 
> The conditions under which one process has permission to obtain the CPU-time clock ID of other processes are implementation-defined.

See https://pubs.opengroup.org/onlinepubs/9699919799/functions/clock_getcpuclockid.html

See RFC #51211 for more info.

jaiiarora, 2023-12-20:
I would like to work on this issue. Can you assign it to me?

cfriedt, 2023-12-20:
> I would like to work on this issue. Can you assign it to me? @cfriedt 

Hi @jaiiarora - typically maintainers get assigned issues, but you are absolutely free to create a pull request to add this feature

Thanks for your help, and I look forward to your PR

jaiiarora, 2023-12-22:
@cfriedt Hi I am trying to implement clock_getcpuclockid() but I could not find **pid_t** or **getpid() function** that are needed for the implementation. Can you suggest me how to tackle this?

cfriedt, 2023-12-22:
> @cfriedt Hi I am trying to implement clock_getcpuclockid() but I could not find **pid_t** or **getpid() function** that are needed for the implementation. Can you suggest me how to tackle this?

I don't think we have `pid_t` or `getpid()` yet.

I do have a draft PR to add them though :-)

In any case, probably the best way to ask for feedback on a PR before it's ready for review is to create a draft PR.

The "Create PR" button in GitHub also has a drop-down menu where you can select "Create Draft PR" instead.

cfriedt, 2023-12-22:
@jaiiarora - I'll separate-out the declaration of `pid_t` and `getpid()` to another PR. That will hopefully unblock you.

jaiiarora, 2023-12-22:
Sure. Thanks a lot @cfriedt

jaiiarora, 2023-12-28:
Hi @cfriedt. I have created a draft PR. Kindly have a look at it. Please let me know if any changes are needed. Thanks

cfriedt, 2023-12-28:
> Hi @cfriedt. I have created a draft PR. Kindly have a look at it. Please let me know if any changes are needed. Thanks

Hi @jaiiarora - are you on discord by any chance? Can you add me?

I would be happy to give you a few pointers "offline" so you can get your draft in better shape before publishing.

jaiiarora, 2023-12-28:
Sure. Let me ping you @cfriedt

jaiiarora, 2023-12-30:
Hi @cfriedt. I have added the requested changes.
- added new macro for CLOCK_PROCESS_CPUTIME_ID
- changed clock_getcpuclockid accordingly
- moved getpid delcaration to zephyr/posix/unistd.h
Please have a look at it and let me know if any changes are needed.

closed 2024-01-08 by carlescufi

jaiiarora, 2024-01-08:
Hi @cfriedt. I have added the missing test case in this [PR.](https://github.com/zephyrproject-rtos/zephyr/pull/67337) Please have a look at it. Thanks

commit a3573a9e (2024-01-08): posix: implement clock_getcpuclockid function

Implements clock_getcpuclockid function

Fixes #59954
PR #67337 (2024-01-20): posix: Adds test for clock_getcpuclockid()
Adds test for clock_getcpuclockid() to
make sure clock_id is equal to CLOCK_PROCESS_CPUTIME_ID after a successful call

Adds test to check getpid() function is not NULL

Fixes #59954
commit b103dece (2024-01-20): posix: Adds test for clock_getcpuclockid()

Adds test for clock_getcpuclockid() to
make sure clock_id is equal to CLOCK_PROCESS_CPUTIME_ID
after a successful call

Fixes #59954