zephyrproject-rtos/zephyr #59958: USB DFU sample gives error : lost device after reset on STM32F429I-DISCO

albayenes, 2023-07-03:
**Describe the bug**

Tried the sample samples/subsys/usb/dfu with mcuboot on a STM32F429I-DISCO board, strictly following the instructions by the README. However, the DFU function gives error : ` Lost device after RESET? ` 

Adding the following solves the problem. However, I am not sure the solution is the best one.

```
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -406,6 +406,7 @@ static int usb_dc_stm32_init(void)
                return -EIO;
        }

+       k_usleep(10);
        /* On a soft reset force USB to reset first and switch it off
         * so the USB connection can get re-initialized
         */

```

**To Reproduce**
Steps to reproduce the behavior:
1. west build <zephyr-workspace>/bootloader/mcuboot/boot/zephyr -d 01-build-mcuboot-STM32f429I_DISC1 -b stm32f429i_disc1
2. west build <zephyr-workspace>/zephyr/samples/subsys/usb/dfu -d 01-build-dfu-STM32f429I_DISC1. -b stm32f429i_disc1
3. <zephyr-workspace>/bootloader/mcuboot/scripts/imgtool.py sign \
        --key  <zephyr-workspace>/bootloader/mcuboot/root-rsa-2048.pem \
        --header-size 0x200 \
        --align 8 \
        --version 1.2 \
        --slot-size 0x60000 \
       <zephyr-workspace>/01-build-dfu-STM32f429I_DISC1/zephyr/zephyr.bin \
       <zephyr-workspace>/01-build-dfu-STM32f429I_DISC1/zephyr/signed-dfu-app.bin
4. openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f4x.cfg
5. telnet 127.0.0.1 4444
6. Using telnet connection format flash using following commands
7. reset
8. halt 
9. stm32f2x mass_erase 0
10. reset
11. halt
12. flash write_image erase <zephyr-workspace>/01-build-mcuboot-STM32f429I_DISC1/zephyr/zephyr.bin 0x08000000
13. flash write_image erase <zephyr-workspace>/01-build-dfu-STM32f429I_DISC1/zephyr/signed-dfu-app.bin 0x0802000
14. exit
15. sudo dfu-util --alt 1  -D <sample-project-build-directory>/zephyr/signed-app.bin

**Expected behavior**
`dfu-util` should correctly upload the firmware image that is currently stored on the device flash.

**Impact**
Prevent my project from enabling the DFU function.

**Logs and console output**
``` 
sudo dfu-util --alt 1  -D <zephyr-<workspace>/01-build-blinky-STM32F429ZI-DISC1/zephyr/signed-app.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 2fe3:0101
Device DFU version 0110
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
dfu-util: Lost device after RESET?

``` 

**Environment (please complete the following information):**
``` 
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.106.0
Qt Version: 5.15.9
Kernel Version: 6.1.31-2-MANJARO (64-bit)
Graphics Platform: X11
Processors: 20 × 12th Gen Intel® Core™ i9-12900H
Memory: 15,4 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: HUAWEI
Product Name: CREF-XX
System Version: M1010
```

**Additional context**
I've found the solution, adding sleep between  HAL_PCD_Init and  HAL_PCD_Stop  solves the problem.

``` 
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -406,6 +406,7 @@ static int usb_dc_stm32_init(void)
                return -EIO;
        }

+       k_usleep(10);
        /* On a soft reset force USB to reset first and switch it off
         * so the USB connection can get re-initialized
         */
```

Pryzy, 2023-09-08:
I had the same issue with Nucleo_F429ZI board.
Following README from amples/subsys/usb/dfu gave the same issue where dfu-util was stuck with "dfu-util: Lost device after RESET?".

Applying the fix listed by @albayenes fixed my issue (though I have also doubts if this solution is the best one - but it makes DFU work).

albayenes, 2023-09-08:
This is not he genuine solution as I remember. Could you share prj.conf?

albayenes, 2023-09-08:
I remember the genuine solution. Please add `CONFIG_USB_DFU_WILL_DETACH=n` to your prj.conf.

Pryzy, 2023-09-08:
@albayenes Indeed adding CONFIG_USB_DFU_WILL_DETACH=n to prj.conf solves this issue without the workaround in usb_dc_stm32.c.

Thanks for update!

closed 2023-09-08 by albayenes