zephyrproject-rtos/zephyr #59984: mbedTLS module does not have CONFIG option for HKDF

asousa-trd, 2023-07-04:
**Describe the bug**
I want to enable mbedTLS HKDF but zephyr module does not have a CONFIG option to enable this.
It is possible to add this one?

The file _zephyr/modules/mbedtls/configs/config-tls-generic.h_ have configs for almost every mbedtls libraries but not for HKDF.
The solution would be to add to _zephyr/modules/mbedtls/Kconfig.tls-generic_ a CONFIG for HKDF.

At this moment, the current solution for me is to define my own MBEDTLS_CFG_FILE, but it would be better to have it in the zephyr.

**To Reproduce**
Steps to reproduce the behavior:
1. Include in one file the `mbedtls_hkdf()` function
2. west build
3. See error: _undefined reference to `mbedtls_hkdf'_

asousa-trd, 2023-07-04:
> This sounds more like an enhancement request than a bug to me?

Yes, you're right. Should I close and open it in enhancement?

henrikbrixandersen, 2023-07-04:
> Yes, you're right. Should I close and open it in enhancement?

No need. It was reclassified as an enhancement.