zephyrproject-rtos/zephyr #59990: Improve coding guidlines descriptions

simhein, 2023-07-04:
The coding guideline rules are described in the documentation under following link:
https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#main-rules

What is missing a description for every Rule why the rule was chosen for the project and what are the benefits for using the rule.

The Task here would be to add a description column to every rule with a description why we using it and what is the benefit of using it.

Coding guideline rules:

- [ ] Dir 1.1
- [ ] Dir 2.1 
- [ ] Dir 3.1
- [ ] Dir 4.1
- [ ] Dir 4.2
- [ ] Dir 4.4
- [ ] Dir 4.5
- [ ] Dir 4.6
- [ ] Dir 4.7
- [ ] Dir 4.8
- [ ] Dir 4.9
- [ ] Dir 4.10
- [ ] Dir 4.11
- [ ] Dir 4.12
- [ ] Dir 4.13
- [ ] Dir 4.14
- [ ] Rule 1.2
- [ ] Rule 1.3
- [ ] Rule 2.1
- [ ] Rule 2.2
- [ ] Rule 2.3
- [ ] Rule 2.6 
- [ ] Rule 2.7
- [ ] Rule 3.1
- [ ] Rule 3.2
- [ ] Rule 4.1
- [ ] Rule 4.2
- [ ] Rule 5.1
- [ ] Rule 5.2
- [ ] Rule 5.3
- [ ] Rule 5.4
- [ ] Rule 5.5
- [ ] Rule 5.6
- [ ] Rule 5.7
- [ ] Rule 5.8
- [ ] Rule 5.9
- [ ] Rule 6.1
- [ ] Rule 6.2
- [ ] Rule 7.1
- [ ] Rule 7.2
- [ ] Rule 7.3
- [ ] Rule 7.4
- [ ] Rule 8.1
- [ ] Rule 8.2
- [ ] Rule 8.3
- [ ] Rule 8.4
- [ ] Rule 8.5
- [ ] Rule 8.6
- [ ] Rule 8.8
- [ ] Rule 8.9
- [ ] Rule 8.10
- [ ] Rule 8.12
- [ ] Rule 8.14
- [ ] Rule 9.1
- [ ] Rule 9.2
- [ ] Rule 9.3
- [ ] Rule 9.4
- [ ] Rule 9.5
- [ ] Rule 10.1
- [ ] Rule 10.2
- [ ] Rule 10.3
- [ ] Rule 10.4
- [ ] Rule 10.5
- [ ] Rule 10.6
- [ ] Rule 10.7
- [ ] Rule 10.8
- [ ] Rule 11.2
- [ ] Rule 11.6
- [ ] Rule 11.7
- [ ] Rule 11.8
- [ ] Rule 11.9
- [ ] Rule 12.1
- [ ] Rule 12.2
- [ ] Rule 12.4
- [ ] Rule 12.5
- [ ] Rule 13.1
- [ ] Rule 13.2
- [ ] Rule 13.3
- [ ] Rule 13.4
- [ ] Rule 13.5
- [ ] Rule 13.6
- [ ] Rule 14.1
- [ ] Rule 14.2
- [ ] Rule 14.3
- [ ] Rule 14.4
- [ ] Rule 15.2
- [ ] Rule 15.3
- [ ] Rule 15.6
- [ ] Rule 15.7
- [ ] Rule 16.1
- [ ] Rule 16.2
- [ ] Rule 16.3
- [ ] Rule 16.4
- [ ] Rule 16.5
- [ ] Rule 16.6
- [ ] Rule 16.7
- [ ] Rule 17.1
- [ ] Rule 17.2
- [ ] Rule 17.3
- [ ] Rule 17.4
- [ ] Rule 17.5
- [ ] Rule 17.6
- [ ] Rule 17.7
- [ ] Rule 18.1
- [ ] Rule 18.2
- [ ] Rule 18.3
- [ ] Rule 18.5
- [ ] Rule 18.6
- [ ] Rule 18.8
- [ ] Rule 19.1
- [ ] Rule 20.2
- [ ] Rule 20.3
- [ ] Rule 20.4
- [ ] Rule 20.7
- [ ] Rule 20.8
- [ ] Rule 20.9
- [ ] Rule 20.11
- [ ] Rule 20.12
- [ ] Rule 20.13
- [ ] Rule 20.14
- [ ] Rule 21.1
- [ ] Rule 21.2
- [ ] Rule 21.3
- [ ] Rule 21.4
- [ ] Rule 21.6
- [ ] Rule 21.7
- [ ] Rule 21.9
- [ ] Rule 21.11
- [ ] Rule 21.12
- [ ] Rule 21.13
- [ ] Rule 21.14
- [ ] Rule 21.15
- [ ] Rule 21.16
- [ ] Rule 21.17
- [ ] Rule 21.18
- [ ] Rule 21.19
- [ ] Rule 21.20
- [ ] Rule 22.1
- [ ] Rule 22.2
- [ ] Rule 22.3
- [ ] Rule 22.4
- [ ] Rule 22.5
- [ ] Rule 22.6
- [ ] Rule 22.7
- [ ] Rule 22.8
- [ ] Rule 22.9
- [ ] Rule 22.10

keith-zephyr, 2024-02-09:
@simhein This seems to follow under the safety working group charter.  Is this something the Safety WG could take on?

simhein, 2024-02-11:
@keith-zephyr I thought about the same as the bot tagged me beside you. Feel free to assign this issue to me.