zephyrproject-rtos/zephyr #59993: STM32H7: Ethernet MAC addresses are not unique enough

spe-ciellt, 2023-07-04:
**Describe the bug**
We have received the first batch of our boards containing an STM32H7. It uses Ethernet and the plan was to use the MAC address extracted from the UUID. But on all devices we have received so far this ends up in the same MAC address.

Looking at the implementation on getting the last 3 bytes for the MAC address it  uses `hwinfo_get_device_id()`. By checking the implementation how it is done it seems to extract the first three bytes from `UUIDWord0()`.  And since all our devices  probably comes from the same production batch this ends up in the same three bytes.

I am not sure if the fix needs to be done in `eth_stm32_hal` or in `hwinfo`.

**To Reproduce**
Needs several different boards from the same production batch. But by reading the code it is kind of obvious. Then use `net iface` and all boards have the same MAC address.

**Expected behavior**
A more unique MAC address per device.

**Impact**
Quite severe. We have to hack around it. Either we update the Ethernet driver locally or we create something that sends a `net_mgmt()` with an updated (correct) MAC address to the Ethernet device at bootup. Last one was suggested in the STM32 channel on Discord by someone having noticed the same problem.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain:  Zephyr SDK 0.16.1
 - Zephyr Version 3.4

**Additional context**
We use an updated copy of the `eth_stm32_hal` since we use a non-default PHY. But that should not make a difference, since most of the code comes from the Zephyr version 3.4 `eth_stm32_hal` driver anyway.

erwango, 2023-07-04:
`hwinfo_get_device_id()` has been updated in https://github.com/zephyrproject-rtos/zephyr/pull/24272.
In this PR word0 and word2 were swapped resulting in extracting word2, described as the lot number to be used, when only few bytes are extracted.

Reverting this change would likely fix the issue described as we can expect word0 (X and Y coordinates) to provide more entropy than word2. Note that only the use of 96 bytes guarantee uniqueness. Using 3 bytes from word0 doesn't guarantee uniqueness but will likely solve most part of the issue.

Note: 
For uniqueness, we'd need to extract the full 96 bits using this API and use a "hash24 algorithm" in eth driver.
There are hash functions available from Zephyr (`include/zephyr/sys/hash_function.h`), they are easily accessible but output 32bits. extracting 3 of the 4 output bytes has no guarantee but could be better than using 3 bytes from word0.

@alexanderwachter any other idea ?

erwango, 2023-07-04:
@rlubos Would you have any recommendations on this topic (regarding the MAC address generation)?

From what I can see most drivers propose a random mac address or device tree based using `local-mac-address`.
None of this method fit for cases such as reported by @spe-ciellt

I guess the point to be able to generate the most unique 3 bytes sequence to complete OUI bytes in the MAC address sounds fairly generic.

rlubos, 2023-07-04:
I don't have any recommendations, but what you propose above for this particular driver sounds perfectly reasonable.

alexanderwachter, 2023-07-04:
We had the same discussion for USB. I also proposed to squeeze out entropy by hashing over the full id. CRC 32 or something like that.

erwango, 2023-07-05:
> We had the same discussion for USB. I also proposed to squeeze out entropy by hashing over the full id. CRC 32 or something like that.

Ok, I'm trying to get internal info on how this 96 bits unique ID is built, if there is any other way to do this.
But I think that any how the best approach would be to get a troncated 24 bits out of a hash32 of the 96 bit unique ID in the end

Desvauxm-st, 2023-07-20:
hello
the RM0433.pdf document give you how the UUID (Device electronic signature) is build for H7 series:

https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

see 61 Device electronic signature
to resume:
Unique device ID register (96 bits) Base address: 0x1FF1 E800

Address offset: 0x00 : Bits 31:0 UID[31:0]: X and Y coordinates on the wafer

Address offset: 0x04 : Bits 31:8 UID[63:40]: LOT_NUM[23:0] 
Lot number (ASCII encoded)
Bits 7:0 UID[39:32]: WAF_NUM[7:0] 
Wafer number (8-bit unsigned number)

Address offset: 0x08 :Bits 31:0 UID[95:64]: LOT_NUM[55:24] 
Lot number (ASCII encoded)

try to use only Address offset: 0x00 and half of Address offset: 0x04 for your 48 bits mac address 

normally little chance of falling back on the same number

erwango, 2023-07-20:
> try to use only Address offset: 0x00 and half of Address offset: 0x04 for your 48 bits mac address

Problem is that 3 first bytes of MAC address is OUI, which is vendor specific. So only the 3 last bytes remain, which we have to pick from the 0x00 offset. Then we should also consider that the 32 bits of this register aren't used as we certainly don't have 2^32 parts per wafer. Rather 10000 max, which could be encoded on 13bits. 
So potentially there are only 13 significant bits out of the 32 from the 0x00 address.
Since we don't know how these 32 bits are constructed if we take only 24 bits out of the 32 bits, potentially we take only 5 significant bits, which can encode only 16 different values.

I'll propose a change using crc32_ieee() applied on the whole 96bits.
Out of the 32 bits CRC, we'll use only 24. By troncating the CRC, we'll loose a bit of the 96bits entropy, but maybe 0.1%, not 25%.
Note: I'm using crc32_ieee() since it is already used in the driver so we'll save few bytes of ROM by reusing it rather than introducing a new hash function.

Desvauxm-st, 2023-07-20:
Can you try the folowing example to check:

(I use samples/hello_word to test)
[main.c.txt](https://github.com/zephyrproject-rtos/zephyr/files/12110357/main.c.txt)

the result is like that

[image]

Desvauxm-st, 2023-07-21:
@spe-ciellt
see after:
drivers: ethernet: generate_mac #60671

spe-ciellt, 2023-07-21:
> @spe-ciellt see after: drivers: ethernet: generate_mac #60671

I have had a look at it and it seems to be much improved. Using the whole UUID and then some hashing on top of that seems like a good solution.

I am currently on vacation and don't have access to the hardware, so I will not be able to perform any practical tests at the moment. But I believe in the update and have no objections to merge it (after all comments have been fixed of course).

Thank you to all involved, I might not have commented but enjoyed following the discussions.

FRASTM, 2023-07-31:
Since the drivers: ethernet: generate_mac https://github.com/zephyrproject-rtos/zephyr/pull/60671 is merged could we close this issue ?

spe-ciellt, 2023-07-31:
> Since the drivers: ethernet: generate_mac #60671 is merged could we close this issue ?

From my point of view, yes.

closed 2023-07-31 by spe-ciellt