zephyrproject-rtos/zephyr #60011: ESP32-C3 USB Serial Not Responding

horakdavid, 2023-07-04:
**Describe the bug**
On ESP32-C3, when there is heavy traffic on ESP's internal serial to usb converter, USB Serial will stop responding and chip can't be flashed without resetting.

Problem occurs in Zephyr 3.2.0 and 3.4.0

**To Reproduce**
Issue can be reproduced using [shell_module](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/shell/shell_module) sample.

Steps to reproduce the behavior:
1. Add `esp32c3_devkitm.overlay` devicetree overlay to the sample with folowing content:
```
/ {
      chosen {
          zephyr,shell-uart = &usb_serial;
      };
  };
  
  &usb_serial {
      status = "okay";
      current-speed = <115200>;
  
      #address-cells = <1>;
      #size-cells = <0>;
};
```

2. `west build -b esp32c3_devkitm`
3. `west flash`
4. Create and run the following python script:
```
import serial
import time

ser = serial.Serial('/dev/ttyACM0', 115200, timeout=1)

while True:
    ser.write("aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa\r\n".encode())
    time.sleep(0.001)
```
5. Open serial port in PuTTY.
6. Serial communication will freeze in few minutes.

**Expected behavior**
Serial communication shouldn't freeze.

**Impact**
ESP32-C3 can't be used without external serial to usb converter.

**Environment**
 - Toolchain: Tried both Zephyr SDK 0.15.0 and Espressif Toolchain

**Additional context**
Echo Bot example is working fine. Seems like the problem occurs when asynchronously transmitting and receiving at the same time.

sylvioalves, 2023-07-21:
@jhedberg i've tested and see no issue at all, only a bunch of `<wrn> shell_uart: RX ring buffer full.` and `aaaaaaaaaa: command not found`.  Also, you requested serial baudrate of 115200bps but is sending 1.08Mbit of data in that python loop.

horakdavid, 2023-07-25:
@sylvioalves What board did you use? On ESP32-C3-DevKitM-1 there are two resistors that needs to be resoldered, otherwise USB is connected to the external serial converter.

sylvioalves, 2023-07-25:
> 

@horakdavid I used esp32-c3-devkitc-02 v1.1. I attached pins 18 and 19 directly into USB port, bypassing devkit's internal USB serial converter. You can easily reproduce this issue?

horakdavid, 2023-07-25:
@sylvioalves I have ESP32-C3-DevKitM-1 v1.0 with resoldered resistors (R1 -> R2 and R4 -> R3), so that the on-board usb connector is connected to pins 18 and 19.

When i run the python script it usually takes about 2-3 minutes till the chip is freezed.

jhedberg, 2023-09-26:
Any update on this one? (it's getting the Stale label added)

horakdavid, 2023-11-27:
Internal usb to serial converter is working fine, the problem was incorrectly set thread priorities

closed 2023-11-27 by horakdavid