zephyrproject-rtos/zephyr #60025: Zephyr 3 .2.99 SPI NOR driver gets wrong flash size

snowuyl, 2023-07-05:
I add printk() in spi_nor_erase() of D:\ncs\v2.3.0\zephyr\drivers\flash\[spi_nor.c](https://devzone.nordicsemi.com/cfs-filesystemfile/__key/communityserver-discussions-components-files/4/6866.spi_5F00_nor.c?_=638241189361271391) for printing flash_size. And it shows flash_size=0x80000 in RTT log messages. But flash_size=0x80000 is same as size = <0x400000> defined in [custom_nrf52832.dts](https://devzone.nordicsemi.com/cfs-file/__key/communityserver-discussions-components-files/4/3348.custom_5F00_nrf52832.dts).

Source code of SPI NOR driver is as follows.

static int spi_nor_erase(const struct device *dev, off_t addr, size_t size)
{
    printk("spi_nor_erase() e\n");
    const size_t flash_size = dev_flash_size(dev);
    int ret = 0;

    /* erase area must be subregion of device */
    if ((addr < 0) || ((size + addr) > flash_size)) {
        printk("addr=0x%lx, size=0x%x, flash_size=0x%x\n", addr, size, flash_size);
        return -ENODEV;
    }

...

}

custom_nrf52832.dts is as follows.

&spi1 {
compatible = "nordic,nrf-spi";
status = "okay";

cs-gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;
pinctrl-0 = <&spi1_default>;
pinctrl-1 = <&spi1_sleep>;
pinctrl-names = "default", "sleep";

w25q32: w25q32jv@0 {
compatible = "jedec,spi-nor";
reg = <0>;
spi-max-frequency = <8000000>;
jedec-id = [ef 40 16];
size = <0x400000>;
has-dpd;
t-enter-dpd = <3500>;
t-exit-dpd = <3500>;
};
};

RTT debug messages are as follows.

w25q32jv@0 SPI flash testing
0> ==========================
0>
0> Test 1: Flash erase
0> spi_nor_erase() e
0> addr=0xff000, size=0x1000, flash_size=0x80000
0> Flash erase failed! -19

snowuyl, 2023-07-05:
[custom_nrf52832.dts.zip](https://github.com/zephyrproject-rtos/zephyr/files/11953308/custom_nrf52832.dts.zip)
[spi_nor.c.zip](https://github.com/zephyrproject-rtos/zephyr/files/11953309/spi_nor.c.zip)
[flash_nor.c.zip](https://github.com/zephyrproject-rtos/zephyr/files/11953310/flash_nor.c.zip)
[custom_nrf52832.zip](https://github.com/zephyrproject-rtos/zephyr/files/11953311/custom_nrf52832.zip)
[minimal_log_flash_erase.zip](https://github.com/zephyrproject-rtos/zephyr/files/11953312/minimal_log_flash_erase.zip)

Laczen, 2023-07-05:
@snowuyl, the size in the spi-nor device is in bit (many producers specify the size in Mbit) instead of the usual byte. There is nothing wrong: 0x400000 bit = 0x80000 byte

closed 2023-07-06 by Laczen