zephyrproject-rtos/zephyr #60029: printk: An exception during spinlock critical area causes a recursive spinlock

povergoing, 2023-07-05:
**Describe the bug**
`CONFIG_PRINTK_SYNC` uses spinlock to do the synchronization.
If an exception, say data abort, happened during the spinlock critical area, the fatal error handler will face a recursive spinlock issue.
Specifically, taking arm64 as an example, 
1. If data abort happened in Line 143~ Line 149 (see following), and now the spinlock is locked.
https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/printk.c#L139-L153
2. Next, the system jumps into the fatal error handler
3. Then the fatal handler wants to `printk` the error information. The `printk` tries to lock the spinlock again and thus a recursive spinlock happens.

[RFC], my solution would be, 
Option 1: detecting in `printk` whether the system is in fatal error mode before the `printk` takes the spinlock, if yes, do not spinlock and just print.
Option 2: fatal error handler uses another (we add a new no lock one) `printk`.

**To Reproduce**
Steps to reproduce the behavior:
1. hack code in Line 143~ Line 149 by injecting a data abort
4. any board can reproduce

**Expected behavior**
no recursive spinlock

**Impact**
showstopper

**Logs and console output**
No logs 

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - main

**Additional context**
I am doing arm64 hardware stack protection, the data abort can happen during printk due to the stackoverflow. The stackoverflow can be detected but not printed because of this recursive spinlock issue.

povergoing, 2023-07-05:
ping @carlocaione FYI

carlocaione, 2023-07-05:
In this case we should add a new `printk_nolock` to be used in all the panic handlers. Also because this is a general problem when panicking while holding a spinlock.

povergoing, 2023-07-05:
> In this case we should add a new `printk_nolock` to be used in all the panic handlers. Also because this is a general problem when panicking while holding a spinlock.

I agree, but the problem, I think, is that the change might affect many things, the error handler is using LOGERR. Do we replace all the LOG_ERR with printk_nolock or ...? Is there anyone I should tag?

carlocaione, 2023-07-05:
@andyross here is your man I think