zephyrproject-rtos/zephyr #60054: PM: pm_suspend_device should be able to handle different states instead of just pushing the devices to suspend mode.

SumitJain3083, 2023-07-05:
Currently the pm_suspend_devices do not take any parameter and by default puts the devices to the SUSPEND mode.
For scenarios like Suspend to RAM , it should be able to take the parameter and call the devices to power off.

It is upon the each individual device if they want to implement the power off or not , but when the system is going to lower power states the devices should be allowed to turn off instead of default suspend mode.

henrikbrixandersen, 2023-07-10:
> How to know if someone is acknowledging this as an request ? 

You only submitted this 5 days ago. You need to be a bit more patient.

> @henrikbrixandersen any help you can do to point this enhancement request to right direction ?

Personally, I think your request makes sense. You could try bringing it up in the Architecture WG on Tuesdays. 

@gmarull Thoughts?

gmarull, 2023-07-12:
To me, we don't need new PM states (it would be a never-ending case), but a call that allows to obtain more info about the suspend process. This way, drivers can take different actions during suspend. We could also introduce a new mechanism to get called during specific events, e.g. s2ram (separate from resume/suspend logic)

SumitJain3083, 2023-07-19:
@gmarull Agreed , my 2 cents on this is , if suspend devices can be parameterized then its fairly easy for each device to understand what state each device is required to go into.

ceolin, 2023-08-11:
@SumitJain3083 have you taken a look in https://github.com/zephyrproject-rtos/zephyr/blob/37cc256006828dd630f5a81abeeabdfb921e5abe/include/zephyr/pm/pm.h#L115 ? Note that from the system perspective the device will be suspended and later we will call to resume it. But it is up to the device to decide what to do.

SumitJain3083, 2023-08-16:
@ceolin  What i want to point out here is , Let's say SoC needs the device to perform different behavior  in different scenario. 

Just For Example 

Scenario 1 :-  Suspend to RAM , Here the device will be Suspended and the configured routine say routine_1 will be called.
Scenario 2 :-  Suspend to DISK , Here the device will be Suspended again and the configured routine routine_1 will be called.

With the current implementation , the devices suspend call can not be configured according to scenarios and it always be called with suspended parameter. 

Can we tackle above situation with the current design ?

gmarull, 2023-08-16:
> @ceolin What i want to point out here is , Let's say SoC needs the device to perform different behavior in different scenario.
>
> Just For Example
>
> Scenario 1 :- Suspend to RAM , Here the device will [...]

You can technically use `pm_state_next_get` inside the PM hooks.

SumitJain3083, 2023-08-21:
> You can technically use `pm_state_next_get` inside the PM hooks.

@gmarull 
or we can put pm_state_next_get as parameter in the "pm_suspend_devices(void)" to let the devices know the state they are supposed to go , instead each device call the API ?

ceolin, 2023-08-23:
> @gmarull or we can put pm_state_next_get as parameter in the "pm_suspend_devices(void)" to let the devices know the state they are supposed to go , instead each device call the API ?

That would impact the device pm callback that is the same one used in device runtime pm. So, changing it would require two separated callbacks or one additional parameter that does not make sense in that usage. Based on current usage, devices that need different behavior depending on the system power state seems to be exceptions and they can inquire which the state the system is transitioning using `pm_state_next_get`. Does it make sense ?