zephyrproject-rtos/zephyr #60065: USB COMPOSITE ACM AND DFU gives error : MPU Fault on STM32F429I-DISCO

albayenes, 2023-07-05:
**Describe the bug**
Setting usb as composite of usb acm and dfu causes MPU fault error during firmware update on STM32F429I-DISCO. To prevent MPU fault, size check is necessary before usb_transfer.

I have one possible solution

```

diff --git a/subsys/usb/device/class/cdc_acm.c b/subsys/usb/device/class/cdc_acm.c
index ab14732440..3cc5bbc319 100644
--- a/subsys/usb/device/class/cdc_acm.c
+++ b/subsys/usb/device/class/cdc_acm.c
@@ -302,7 +302,8 @@ static void cdc_acm_read_cb(uint8_t ep, int size, void *priv)
        }

done:
-       usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
+       if (size > 0)
+               usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
                     USB_TRANS_READ, cdc_acm_read_cb, dev_data);
}

```

**To Reproduce**

DTS file

```

/*
 * Copyright (c) 2017 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/dts-v1/;
#include <st/f4/stm32f429Xi.dtsi>
#include <st/f4/stm32f429zitx-pinctrl.dtsi>
#include <zephyr/dt-bindings/display/ili9xxx.h>

/ {
	model = "STMicroelectronics STM32F429I_DISC1 board";
	compatible = "st,stm32f4discovery";

	chosen {
		zephyr,code-partition = &slot0_partition;
		zephyr,console = &usart1;
		zephyr,shell-uart = &usart1;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,ccm = &ccm0;
		zephyr,display = &ili9341;
	};

	sdram2: sdram@d0000000 {
		compatible = "zephyr,memory-region", "mmio-sram";
		device_type = "memory";
		reg = <0xd0000000 DT_SIZE_M(8)>;
		zephyr,memory-region = "SDRAM2";
	};

	leds {
		compatible = "gpio-leds";
		orange_led_3: led_3 {
			gpios = <&gpiog 13 GPIO_ACTIVE_HIGH>;
			label = "User LD3";
		};
		green_led_4: led_4 {
			gpios = <&gpiog 14 GPIO_ACTIVE_HIGH>;
			label = "User LD4";
[... 299 lines cut ...]
Memory: 15,4 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: HUAWEI
Product Name: CREF-XX
System Version: M1010

**Additional context**

I've found the solution, adding sleep between HAL_PCD_Init and HAL_PCD_Stop solves the problem.

```
diff --git a/subsys/usb/device/class/cdc_acm.c b/subsys/usb/device/class/cdc_acm.c
index ab14732440..3cc5bbc319 100644
--- a/subsys/usb/device/class/cdc_acm.c
+++ b/subsys/usb/device/class/cdc_acm.c
@@ -302,7 +302,8 @@ static void cdc_acm_read_cb(uint8_t ep, int size, void *priv)
        }

done:
-       usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
+       if (size > 0)
+               usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
                     USB_TRANS_READ, cdc_acm_read_cb, dev_data);
}
```

tmon-nordic, 2023-07-06:
> I have one possible solution
>
> ```
>
> diff --git a/subsys/usb/device/class/cdc_acm.c b/subsys/usb/device/class/cdc_acm.c
> index ab14732440..3cc5bbc319 100644
> --- a/subsys/usb/device/class/cdc_acm.c
> +++ [...]

But this would kill the IN endpoint, wouldn't it? Are you able to receive anything after the `if (size > 0)` branch is not taken?

albayenes, 2023-07-06:
> > I have one possible solution
> > ```
> > 
> > diff --git a/subsys/usb/device/class/cdc_acm.c b/subsys/usb/device/class/cdc_acm.c
> > index ab14732440..3cc5bbc319 100644
> > --- a/subsys/usb/device/class/cdc_acm.c
> > +++ b/subsys/usb/device/class/cdc_acm.c
> > @@ -302,7 +302,8 @@ static void cdc_acm_read_cb(uint8_t ep, int size, void *priv)
> >         }
> > 
> > done:
> > -       usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
> > +       if (size > 0)
> > +               usb_transfer(ep, dev_data->rx_buf, sizeof(dev_data->rx_buf),
> >                      USB_TRANS_READ, cdc_acm_read_cb, dev_data);
> > }
> > ```
> 
> But this would kill the OUT endpoint, wouldn't it? Are you able to receive anything after the `if (size > 0)` branch is not taken?

Yes you are right. I am working on it. I have to fins a way to prevent usb transfer during DFU communication. Are there such flags? Is it valid usb transfer during DFU update?

tmon-nordic, 2023-07-06:
> Yes you are right. I am working on it. I have to find a way to prevent usb transfer during DFU communication. Are there such flags? Is it valid usb transfer during DFU update?

The is what the Run-Time and DFU Mode are for in the DFU Specification. In Run-Time mode, the host executes DFU_DETACH and then device switch to DFU Mode (enumerates again on the bus exposing different configuration and different product ID).

USB Device Firmware Upgrade Specification, Revision 1.1
> 4.2.2 DFU Mode Configuration Descriptor
> This descriptor is identical to the standard configuration descriptor described in the USB specification
> version 1.0, with the exception that the bNumInterfaces field must contain the value 01h.

This basically means you cannot have CDC ACM simultaneously when device is in DFU Mode. You can only have simultaneous Run-Time and CDC ACM.

The first thing to check would be to determine whether DFU_DETACH correctly frees all resources on your target.

albayenes, 2023-07-06:
> The is what the Run-Time and DFU Mode are for in the DFU Specification. In Run-Time mode, the host executes DFU_DETACH and then device switch to DFU Mode (enumerates again on the bus exposing [...]

As far as I understand, ` usb_dc_detach ` in usb_dc_stm32.c is not working correctly. Do you think this is the possible reason?

tmon-nordic, 2023-07-06:
> As far as I understand, `usb_dc_detach` in usb_dc_stm32.c is not working correctly. Do you think this is the possible reason?

Yes if `CONFIG_USB_DFU_WILL_DETACH` is enabled. You can disable `USB_DFU_WILL_DETACH` in Kconfig and check.

albayenes, 2023-07-06:
> USB_DFU_WILL_DETACH

Disabling CONFIG_USB_DFU_WILL_DETACH solves the problem. Thanks for your help. But now gives following error

```
Waiting for device, exit with ctrl-C
Opening DFU capable USB device...
Device ID 2fe3:0001
Run-Time device DFU version 0110
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Resetting USB...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 128
Copying data from PC to DFU device
Download        [==                       ]   8%        21376 bytesdfu-util: Error during download (LIBUSB_ERROR_TIMEOUT)

```

albayenes, 2023-07-06:
> Disabling CONFIG_USB_DFU_WILL_DETACH solves the problem. Thanks for your help. But now gives following error
>
> ```
> Waiting for device, exit with ctrl-C
> Opening DFU capable USB device...
> Device ID [...]

Increasing stack sizes solves the problem.

```

CONFIG_ISR_STACK_SIZE=9096

CONFIG_IDLE_STACK_SIZE=9096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=9096
CONFIG_USB_WORKQUEUE_STACK_SIZE=9096

CONFIG_PRIVILEGED_STACK_SIZE=9096

```

closed 2023-07-06 by albayenes