zephyrproject-rtos/zephyr #60070: Add klocwork support in Zephyr 3.0.99

shubhamtotade, 2023-07-06:
Hi @marc-hb
I am working on project with zephyr 3.0.99
So though of cherrypicking these 4 commits from #52671 and port it to my project and modify sparse support to klocwork support
1. [cmake: implement build infrastructure for supporting SCA tools.] - As it is
2. [cmake: add sparse support to the new SCA infrastructure]  - Made some changes In cmake/sca/klocwork/sca.cmake.
Made some changes in klocwork.template. 

But while compiling my project its giving error 

```
Build files have been written to: /workdir/cpucp.proc

west build: building application

[1/245] Preparing syscall dependency handling

[2/245] Generating include/generated/version.h

Zephyr version: 3.0.99 (/workdir/zephyr), build: ve.2.51 [8/245) Building C object zephyr/CMakeFiles/offsets.dir/arch/riscv/core/offsets/offsets.c.obj cmake -E env: unknown option -DCPUCP_MODULE_INIT_LEVEL-POST KERNEL'

[9/245] Generating include/generated/offsets.h FAILED: zephyr/include/generated/offsets.h

cd /workdir/cpucp_proc/zephyr && /usr/bin/python3.8 /workdir/zephyr/scripts/gen_offset_header.py -1/workdir/cpucp_proc/zephyr/CMakeFiles/offsets.dir/./arc h/riscv/core/offsets/offsets.c.obj -o /workdir/cpucp_proc/zephyr/include/generated/offsets.h

Traceback (most recent call last):

File "/workdir/zephyr/scripts/gen_offset_header.py", line 80, in <module>

input file = open(args.input, 'rb')

FileNotFoundError: [Errno 2] No such file or directory: '/workdir/cpucp_proc/zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj'

ninja: build stopped: subcommand failed.

FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build/workdir/cpucp.proc

caught signal:
```

_Originally posted by @shubhamtotade in https://github.com/zephyrproject-rtos/zephyr/issues/52671#issuecomment-1622159545_

nordicjm, 2023-07-06:
Is this a bug, feature request, rfc, binary blob request, etc.? Can you update this with the required zephyr template details filled in?

shubhamtotade, 2023-07-07:
Its a feature request. 
Let me know where can I find the template so that I can fill it

nordicjm, 2023-07-07:
I've put the template in your original post, please edit and fill it in

closed 2023-07-07 by shubhamtotade