zephyrproject-rtos/zephyr #60074: testsuite: ztest_new:

MarekPieta, 2023-07-06:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2023-07-17:
@yperess  related to shuffling features, can you please take a look?

fabiobaltieri, 2023-07-25:
@yperess ping

asemjonovs, 2023-07-25:
I took a look at this and it looks like we reset the test statistics on the next iteration of test or suite leading to inaccurate results displayed.  In this scenario this specific test will fail 1 out of 3 times.  Since the last two were passing, it shows `SUITE PASSED`.  If you reduce the shuffle suite/test counts to 1, you won't observe this.  I think if we should display a new test result of `FLAKY` if a fraction of the tests pass.  `PASS` when all pass, or `FAIL` when all failed.  My bandwidth is a bit limited at the moment to work on this. 

@smrtos - I think you added the test result summary initially.  Do you have any cycles to work on this?

nashif, 2023-08-02:
> @smrtos - I think you added the test result summary initially. Do you have any cycles to work on this?

@asemjonovs  does not work on Zephyr anymore.