zephyrproject-rtos/zephyr #60085: PM: Define the IDLE function as weak symbol.

SumitJain3083, 2023-07-06:
Currently the Zephyr framework mandates to use its own implementation of IDLE framework which is tied up with system specific implementation of policy and state manager. 

However there is no flexibility in following the idle framework. if we can manage to define the idle function as weak , then SOC specific framework can be implemented according to the SOC needs.

Please evaluate if idle function can be defined as weak symbol ?

gmarull, 2023-07-12:
We have better alternatives than weaks (non standard), that is, Kconfig to select upon multiple options. Said that, re-implementing the whole idle thread seems like a risky choice nowadays. Maybe adding support for certain hooks is a better idea.

ceolin, 2023-08-11:
@SumitJain3083 Can you describe the use case ? There isn't much things done by the idle thread that are not customizable. It basically call arch specific code to gets idle (that can be overwritten) or triggers the power management subsystem that is pretty customizable. So, would be nice to understand what you are trying to do.

SumitJain3083, 2023-08-11:
@ceolin  I find the implementation of idle function pretty decent when it comes to calling the arch specific code to call k_cpu_idle() but the moment you enable the flag to call pm_system_suspend() then there are couple of possible options that SOC want to do. 

1. Going for deeper power states and for each state incrementally calling the suspend on the devices.
  That is, Say the system is going in RUNTIME IDLE State and SOC needs some devices to turn off , we can't really do that. it 
  has to be all the devices on which Suspend will be called and that too when the state is not RUNTIME IDLE. 

While i am not opposing the current design , i am proposing a flexibility if SoC needs some way of implementing the Sleep according to their needs then they should be allowed to do that.

ceolin, 2023-08-11:
> @ceolin I find the implementation of idle function pretty decent when it comes to calling the arch specific code to call k_cpu_idle() but the moment you enable the flag to call pm_system_suspend() then there are couple of possible options that SOC want to do.

Great, so we can focus in the pm subsystem.

> 1. Going for deeper power states and for each state incrementally calling the suspend on the devices.
>    That is, Say the system is going in RUNTIME IDLE State and SOC needs some devices to turn off [...]

Yep, that is reasonable. There are working happening around this. In this pr (https://github.com/zephyrproject-rtos/zephyr/pull/60939) I propose be able to set which states should trigger power management. That brings more flexibility but still does not solve the problem of which devices should be suspended in each power state. There is another pr (https://github.com/zephyrproject-rtos/zephyr/pull/60998) proposing new points for power management notifiers that devices could use. Though that requires more work in the device itself.

ceolin, 2024-05-07:
Closing it based on discussion. Feel free to open other issue.

closed 2024-05-07 by ceolin