zephyrproject-rtos/zephyr #60096: native drivers and backends not yet supported with embedded C libraries

aescolar, 2023-07-06:
This is just to keep track of the progress for native (posix/sim) drivers which are not yet suported with the embedded C libraries:

* bluetooth userchan
* can can native posix
* entropy - https://github.com/zephyrproject-rtos/zephyr/pull/60033 
* eprom eprom emulator
* ethernet eth native_posix
* flash flash simulator - https://github.com/zephyrproject-rtos/zephyr/pull/60034 
* flash host based flash access
* SDL (GPIO, display, input) - https://github.com/zephyrproject-rtos/zephyr/pull/60036
* serial uart native posix/PTTY - https://github.com/zephyrproject-rtos/zephyr/pull/60053
* serial uart native TTY
* tracing - Posix tracing backend
* usb - USB native posix
* ztests (extra host based functionality)- https://github.com/zephyrproject-rtos/zephyr/pull/60091

CC: @jhedberg @MarkoSagadin @henrikbrixandersen @martinjaeger @tmon-nordic @jfischer-no @finikorg @jukkar @tbursztyka @nvlsianpu @vanwinkeljan @de-nordic @nordic-krch @Laczen

tmon-nordic, 2023-07-07:
I don't think the porting of USB native posix makes sense - it has broken design and does not really implement what USBIP expects. We could however, aim with the new stack (device-next) USBIP implementation to be supported with embedded C libraries.

aescolar, 2023-07-07:
@tmon-nordic Up to you guys (USB maintainers).

aescolar, 2025-03-10:
All but the USB driver (which is meant to be removed) are now buildable with other libCs => closing as done.

closed 2025-03-10 by aescolar