zephyrproject-rtos/zephyr #60103: native_sim_64 build error

alpsayin, 2023-07-06:
**Describe the bug**
error: return type of ‘main’ is not ‘int’ [-Werror=main]

As smoked by CI here: https://github.com/zephyrproject-rtos/zephyr/actions/runs/5474126763/jobs/9968549840?pr=52208#step:13:314
Found as a part of PR https://github.com/zephyrproject-rtos/zephyr/pull/52208

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout PR 52208
2. `west build -p -b native_sim_64 -T samples/coop_synchronization/sample.kernel.coop_synchronization.polled_uart`

**Expected behavior**
Build success.

**Impact**
Annoyance: CI fails.

**Logs and console output**
```
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /usr/bin/gcc ... CMakeFiles/app.dir/src/main.c.obj -c /__w/zephyr/zephyr/samples/coop_synchronization/src/main.c
/__w/zephyr/zephyr/samples/coop_synchronization/src/main.c:117:6: error: return type of ‘main’ is not ‘int’ [-Werror=main]
  117 | void main(void)
      |      ^~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - Zephyr CI
 - Linux env
 
**Additional context**
None

alpsayin, 2023-07-06:
CC @aescolar

aescolar, 2023-07-06:
The bug is in the PR itself. That error will trigger with all target boards, not just native_sim.
Closing as not a bug.

closed 2023-07-06 by aescolar