zephyrproject-rtos/zephyr #60128: eth_mcux: ipv6 works on both interfaces only when promiscuous mode is enabled.

jameswalmsley-cpi, 2023-07-07:
When both interfaces are enabled. I can only ping a single interface unless promiscuous mode is enabled.

jhedberg, 2023-09-12:
@danieldegrasse any comment on this one?

danieldegrasse, 2023-09-20:
> @danieldegrasse any comment on this one?

Unfortunately I don't have a good way to reproduce this, as none of the NXP EVKs I'm aware of have multiple ENET interfaces enabled in Zephyr. @jameswalmsley-cpi, if you can provide more details on where you see the error I can attempt to help narrow this down, but unfortunately I don't have hardware I can reproduce it with.

decsny, 2024-04-24:
This bug report is unclear and seems to be raised against the reporter's own PR that was never merged https://github.com/zephyrproject-rtos/zephyr/pull/60073, so closing this for now, reopen if necessary

closed 2024-04-24 by decsny