zephyrproject-rtos/zephyr #60132: LE Audio: Controller: Failed to establish a second CIS with Packetcraft controller

mkasenberg, 2023-07-07:
**Describe the bug**
The PTS tests with 2 CIS fails because the second CIS could not be established. Looking for an explanation I tested samples/bluetooth/unicast_audio_client app with samples/bluetooth/unicast_audio_server app on nrf5340dk boards, with 2 bidirectional CISes configuration, switching between Zephyr and Packetcraft controller.

There is no problem with second CIS when Zephyr controller is on both sides of a connection. But in Zephyr <-> Packetcraft connection the second CIS always fails to be established. Is there a known solution for this issue?

**Btmon logs**
[unicast_client_zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/11977129/unicast_client_zephyr.log)
[unicast_server_packetcraft.log](https://github.com/zephyrproject-rtos/zephyr/files/11977130/unicast_server_packetcraft.log)

**To Reproduce**
**Unicast Client: Zephyr Host + Zephyr Controller:**
In samples/bluetooth/hci_rpmsg, build the netcore binary:

`west build -b nrf5340dk_nrf5340_cpunet -- -DOVERLAY_CONFIG=nrf5340_cpunet_iso-bt_ll_sw_split.conf`

Flash the controller with:
`west flash --recover`

In samples/bluetooth/unicast_audio_client build the appcore binary:
[unicast_client.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11979294/unicast_client.config.txt)
`west build -b nrf5340dk_nrf5340_cpuapp . -- -DOVERLAY_CONFIG=unicast_client.config.txt`

Flash the app with:
`west flash`

**Unicast Server: Zephyr Host + Packetcraft Controller:**
Download the controller binary:
https://github.com/nrfconnect/sdk-nrf/blob/v2.3.0/applications/nrf5340_audio/bin/ble5-ctr-rpmsg_3330.hex

and flash with:
`nrfjprog --program *.hex --chiperase --coprocessor CP_NETWORK -r`

In samples/bluetooth/unicast_audio_server build the appcore binary:
[unicast_server.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11979303/unicast_server.config.txt)
`west build -b nrf5340dk_nrf5340_cpuapp . -- -DOVERLAY_CONFIG=unicast_server.config.txt`

Flash the app with:
`west flash`

mkasenberg, 2023-12-19:
Not stale at all. I still see some sync timeouts at establishing the second CIS and don't now how to fix this. It affects multiple BAP and CAP test cases when testing with the PTS.

mkasenberg, 2023-12-28:
Affected PTS test cases:
BAP/UCL/STR/BV-525-C
BAP/UCL/STR/BV-526-C
BAP/UCL/STR/BV-527-C
BAP/UCL/STR/BV-528-C
BAP/UCL/STR/BV-529-C
BAP/UCL/STR/BV-531-C
BAP/UCL/STR/BV-532-C
BAP/UCL/STR/BV-533-C
BAP/USR/STR/BV-362-C
BAP/USR/STR/BV-365-C
BAP/USR/STR/BV-366-C
BAP/USR/STR/BV-377-C
BAP/USR/STR/BV-380-C
BAP/USR/STR/BV-383-C

cvinayak, 2024-02-02:
@mtpr-ot and @thoh-ot could please take a look at the attached sniffer logs. unicast_audio_client sample with pcft controller gets back  REJECT IND with reason CIS Accept Timeout from Zephyr Controller as the ISO peripheral (unicast_audio_server)...is there anything you can identify in the Zephyr Controller ULL LLCP implementation that can explain this? (same unicast_audio_client with zephyr controller works fine). This issue is apparently blocking PTS testing.

[pcft_zll_unicast_audio.zip](https://github.com/zephyrproject-rtos/zephyr/files/14140696/pcft_zll_unicast_audio.zip)

cvinayak, 2024-02-08:
@erbr-ot may be your can spot something in Peripheral ISO LLCP, why second CIS_REQ does not generate the notification but is transitioning to waiting for accept and eventually timeout reject ext ind is generated.

thoh-ot, 2024-02-09:
@cvinayak I took a look, but cannot find anything obvious. There are alot if if-checks in the accept path and without a HCI log it is hard to say what is going on.

cvinayak, 2024-02-09:
> @cvinayak I took a look, but cannot find anything obvious. There are alot if if-checks in the accept path and without a HCI log it is hard to say what is going on.

I am continuing with investigation, at this point i am seeing anything wrong with the LLCP. REJECT_EXT_IND was due to application crash (usage fault) in my build of the unicast_audio_server, and CIS_REQ not being accepted/rejected from host due to it having crashed.

cvinayak, 2024-02-09:
@mkasenberg I have been debugging and noticed that I had to workaround the instant used in CIS_IND to be one ahead so that packetcraft controller can establish a peripheral CIS, this is with version 3424. With the same workaround in zephyr controller the packetcraft controller version 3393 sometimes hangs/unresponsive when the second CIS_IND is sent to it.

Is it possible to collect on-air trace when the PTS tests is failing due to the second CIS not being established?

Could you check if the PTS dongle has a newer firmware available?

mkasenberg, 2024-02-12:
The PTS Firmware Upgrade Software shows that the 13.23250.0.0 as the latest version of the firmware. This version has been in use for a couple of months, so for sure it does not use the latest Packetcraft.

mkasenberg, 2024-02-12:
Ellisys sniffer logs
[Uploading BAP_UCL_STR_BV_527_C.btt.txt…]()

cvinayak, 2024-02-12:
The second CIS establishment failure should be worked around in the PR: https://github.com/zephyrproject-rtos/zephyr/pull/68227, as I managed to get the unicast_audio_client sample with ZLL to establish both CISes with the unicast_audio_server with packetcraft controller.

> Ellisys sniffer logs [BAP_UCL_STR_BV_527_C.btt.txt](https://github.com/zephyrproject-rtos/zephyr/files/14236590/BAP_UCL_STR_BV_527_C.btt.txt) PTS logs: [BAP_UCL_STR_BV_527_C_2024_02_12_09_59_20.zip](https://github.com/zephyrproject-rtos/zephyr/files/14236651/BAP_UCL_STR_BV_527_C_2024_02_12_09_59_20.zip)

@mkasenberg I believe this log is for the zephyr controller built from upstream main? Is the logs available for the failure here: https://github.com/zephyrproject-rtos/zephyr/pull/68227#issuecomment-1936441319 ?

mkasenberg, 2024-02-12:
Logs for https://github.com/zephyrproject-rtos/zephyr/pull/68227#issuecomment-1938614064
Ellisys logs:
[BAP_UCL_STR_BV_525_C.btt.txt](https://github.com/zephyrproject-rtos/zephyr/files/14238520/BAP_UCL_STR_BV_525_C.btt.txt)
PTS logs:
[BAP_UCL_STR_BV_525_C_2024_02_12_13_42_56.zip](https://github.com/zephyrproject-rtos/zephyr/files/14238522/BAP_UCL_STR_BV_525_C_2024_02_12_13_42_56.zip)

cvinayak, 2024-02-13:
unicast_audio_client with ZLL
ZLL->SDC: Good
ZLL->PCFT 3424: Good
ZLL->PTS test suite: Bad

[zll_sdc_pcft3424_unicast_audio_good.zip](https://github.com/zephyrproject-rtos/zephyr/files/14260453/zll_sdc_pcft3424_unicast_audio_good.zip)

mkasenberg, 2024-02-13:
I will ask the PTS support if they have any upcoming firmware updates.

mringwal, 2024-02-13:
Related to the PTS firmware: the current one supports Unicast Audio to an up-to-date Pixel 7a, but not to the Pixel 8. The slightly newer PacketCraft LL on the nRF5340 ADK works with the Pixel 8, too.

cvinayak, 2024-02-13:
> unicast_audio_client with ZLL ZLL->SDC: Good ZLL->PCFT 3424: Good ZLL->PTS test suite: Bad
> 
> [zll_sdc_pcft3424_unicast_audio_good.zip](https://github.com/zephyrproject-rtos/zephyr/files/14260453/zll_sdc_pcft3424_unicast_audio_good.zip)

unicast_audio_client with ZLL
ZLL->SDC: Good
ZLL->PCFT 3424: Good
ZLL->PTS test suite: Bad

unicast_audio_client with SDC (nRF Connect SDK):
SDC->PCFT 3424: Bad

Sniffer logs:
[zll_sdc_pcft3424_unicast_audio.zip](https://github.com/zephyrproject-rtos/zephyr/files/14261581/zll_sdc_pcft3424_unicast_audio.zip)

mkasenberg, 2024-02-13:
I've got a response from PTS support and they are waiting for a new version of Packetcraft too, the release date is unknown.

mringwal, 2024-02-16:
Thanks. There's no point reporting the incompatibility with Pixel 8 then, either.

Maybe related to the actual issue: all BAP tests pass using the PTS dongle (slightly older PacketCraft LL) incl. the 2 bidirectional CIS.So ZLL <-> ZLL works as well as PTS PacketCraft <-> PTS PacketCraft, but not when 'mixed'.

mringwal, 2024-02-28:
> all BAP tests pass using the PTS dongle (slightly older PacketCraft LL) incl. the 2 bidirectional CIS.So ZLL <-> ZLL works as well as PTS PacketCraft <-> PTS PacketCraft, but not when 'mixed'.

I'm sorry, this wasn't correct. Setting up two CIS from a PTS dongle to two other PTS dongles fails - while the first CIS get's established, the second does not. 

For comparison, using the newer PacketCraft LL from the ADK on the nRF5340 instead of the PTS dongle works here (i.e. both CIS get established correctly).

cvinayak, 2024-03-15:
@mkasenberg Can we close this for now (and re-open in the future)?, as the issue needs new PTS firmware, and the issue for now is not on the Zephyr Controller side.

MicImbriani, 2024-06-24:
@mkasenberg is there any update on this PTS ticket?

closed 2024-07-08 by Thalley