zephyrproject-rtos/zephyr #60135: Compiling hci_rpmsg for nRF5340 fails on some boards (e.g. in-tree bl5340_dvk_cpunet)

jakkra, 2023-07-07:
**Describe the bug**
When upgrading from Zephyr 3.3 -> 3.4 (and latest main) the Bluetooth hci_rpmsg does not work anymore (for some boards) (with LL_SPLIT .overlays) and lot's of Kconfig warning are generated. It works well with `nrf5340dk_nrf5340_cpunet` but not with our custom boards, which means something probably changed in board files I assume. 

Our board files are not upstream, **but** I could reproduce with the upstream bl5340_dvk_cpunet board. 

I have tried diffing and see what's different and what's missing, but I cannot find it. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
nRF5340
 - What have you tried to diagnose or workaround this issue?
 Diffing board files `bl5340_dvk_cpunet` and `nrf5340dk_nrf5340_cpunet`
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 Between Zephyr 3.3 and 3.4

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/bluetooth/hci_rpmsg
2. west build -p -b bl5340_dvk_cpunet -- -DEXTRA_CONF_FILE=nrf5340_cpunet_iso-bt_ll_sw_split.conf
3. Many BT Controller kconfig warnings and errors

**Expected behavior**
No build errors

**Impact**
Cannot upgrade to Zephyr 3.4 or main or nRF Connect SDK 2.4

**Logs and console output**
[bl5340_dvk_cpunet_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/11980184/bl5340_dvk_cpunet_log.txt)

**Environment (please complete the following information):**
 - OS: Windows, Linux
 - Toolchain: ncs/toolchains/31f4403e35
 - Commit SHA or Version used: cc67aaf207f0979c2a3cd0d7d642310958640373

nordicjm, 2023-07-07:
I would assume it is caused by either rtc0 or timer0 being enabled when it must be disabled, as per:
```
        depends on !$(dt_nodelabel_enabled,timer0)
        depends on !$(dt_nodelabel_enabled,rtc0)
```

nordicjm, 2023-07-07:
https://github.com/zephyrproject-rtos/zephyr/commit/369ffb4929cd5a0f2253e0fdaf15b70ec275c3aa is probably offending commit

nordicjm, 2023-09-06:
Timers should be removed unless your application is using them, so if you want to use e.g. `timer2` then you should configure that, otherwise it would be disabled.

closed 2023-09-06 by jakkra