zephyrproject-rtos/zephyr #60141: ESP32 external ram not utilized

CT3, 2023-07-07:
**Describe the bug**
When building with or without external ram, iram and dram fills up an er_ram in never used. Enabling ble and mcuboot takes up all the ram available, but even having 2mb external ram makes no difference.

**To Reproduce**
run west build -p -b esp32 ./zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DEXTRA_CONF_FILE=overlay-bt.conf

add 
CONFIG_ESP_SPIRAM=y
CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD=1024
CONFIG_HEAP_MEM_POOL_SIZE=98304

to the prj.conf and run again

**Expected behavior**
some of the ram allocation should go to external ram

**Impact**
just a basic smp_svr example is using 97% of ram, so there is no chance to make actual app on top with only 3% ram available

**Logs and console output**
[289/289] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:      351504 B    4194240 B      8.38%
     iram0_0_seg:       75692 B        76 KB     97.26%
     irom0_0_seg:      220496 B         4 MB      5.26%
     dram0_0_seg:      115572 B     124580 B     92.77%
     dram0_1_seg:       92800 B     110032 B     84.34%
     drom0_0_seg:       29464 B    4194240 B      0.70%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
     ext_ram_seg:           4 B         2 MB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%

**Environment (please complete the following information):**
 - OS: ( Windows)
 - Toolchain (Zephyr SDK)
 - Version used 3.4.99

sylvioalves, 2023-07-07:
Hi, I guess you can decrease total HEAP to `CONFIG_HEAP_MEM_POOL_SIZE=40000`.  

About SPIRAM usage:

- It can only start with .bss and .noinit, however, _this is not implemented yet. Will be added as a Kconfig_.
- If you are using network, you can add `CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y` to move network stuff to external ram.

CT3, 2023-07-10:
im using bluetooth only in this example so CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y does nothing.
but changing CONFIG_HEAP_MEM_POOL_SIZE=40000 still compiles with 97% ram usage

sylvioalves, 2023-07-21:
@CT3, sorry for the delay. We have been improving mcuboot integration and this might give you some room in there:  
Would you change the line below to `0x20000` instead?  

`#define IRAM_SEG_LEN 0x13000` to `#define IRAM_SEG_LEN 0x20000`
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/xtensa/esp32/default.ld#L47  

That will free up some IRAM region. This is an ongoing analysis we have been doing though.

rftafas, 2023-08-15:
@CT3 Can you check during boot which revision of ESP32 you are using?

closed 2023-10-31 by github-actions[bot]

Jellearends, 2024-03-11:
For me it worked by adding the following kconfig settings to my prj.conf:
CONFIG_ESP_SPIRAM=y
CONFIG_SPIRAM_TYPE_ESPPSRAM64=y
CONFIG_SPIRAM_MODE_QUAD=y

I hope this helps you as this issue has stumbled me for quite some time