zephyrproject-rtos/zephyr #60145: posix: rwlock: Racy issue causing reader_active sem never been released

povergoing, 2023-07-07:
**Describe the bug**
Three threads unlock simultaneously, `k_sem_count_get(&rwlock->rd_sem)`  might be `CONCURRENT_READER_LIMIT - 3` from all 3 threads view, instead of the expected `CONCURRENT_READER_LIMIT - 1` for the last one. Thus no thread will be identified as the last one to do `k_sem_give(&rwlock->reader_active); ` As a result, the test case hangs.

https://github.com/zephyrproject-rtos/zephyr/blob/b95cdb2a334b380a384be6d1fd042c8312f346b9/lib/posix/rwlock.c#L201-L210

**To Reproduce**
Steps to reproduce the behavior:
1. testcase: tests/posix/common/src/posix_rwlock.c
2. west build -p always -b "any smp board" -t run -T zephyr/tests/posix/common/portability.posix.common.picolibc
3. might need to repeat many times

**Expected behavior**
PASS

**Impact**
showstopper

**Logs and console output**
```
===================================================================
START - test_posix_rw_lock
main acquire WR lock and 3 threads acquire RD lock
Thread 2 scheduling policy = 2 & priority 3 started
Not able to get RD lock on trying, try again
Thread 1 scheduling policy = 2 & priority 2 started
Not able to get RD lock on trying, try again
Thread 0 scheduling policy = 2 & priority 1 started
Not able to get RD lock on trying, try again
Parent thread releasing WR lock
Thread 2 got RD lock
Thread 1 got RD lock
Thread 0 got RD lock
Parent thread acquiring WR lock again
Thread 2 releasing RD lock
Thread 1 releasing RD lock
Thread 2 acquiring WR lock
Thread 0 releasing RD lock
Thread 0 acquiring WR lock
Thread 1 acquiring WR lock
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK

**Additional context**
Not sure if it is the right solution, but solves the issue.
Swap the `k_sem_give` sequence, see below. Although it's possible that all 3 threads will run `k_sem_give(&rwlock->reader_active);` 3 times, the `k_sem_give` will make sure only one takes effect.
```
diff --git a/lib/posix/pthread_rwlock.c b/lib/posix/pthread_rwlock.c
index 45d4af6945..60039fb088 100644
--- a/lib/posix/pthread_rwlock.c
+++ b/lib/posix/pthread_rwlock.c
@@ -200,13 +200,13 @@ int pthread_rwlock_unlock(pthread_rwlock_t *rwlock)
                k_sem_give(&rwlock->wr_sem);
        } else {
                /* Read unlock */
+               k_sem_give(&rwlock->rd_sem);
+
                if (k_sem_count_get(&rwlock->rd_sem) ==
-                                   (CONCURRENT_READER_LIMIT - 1)) {
+                                   CONCURRENT_READER_LIMIT) {
                        /* Last read lock, unlock writer */
                        k_sem_give(&rwlock->reader_active);
                }
-
-               k_sem_give(&rwlock->rd_sem);
        }
        return 0;
 }
```

cfriedt, 2023-07-08:
@povergoing - would you be good to submit a PR?

povergoing, 2023-07-09:
> @povergoing - would you be interested in submitting a PR? Solution looks reasonable to me.

Sure, I will

PR #60231 (2023-07-12): lib: posix: pthread_rwlock: Fix the racy issue causing writer to hang
Fix #60145

Multiple reader threads unlocking the read lock simultaneously might cause the program to hang because it's possible that no thread is identified as the last one to activate the writer thread.

To fix the issue, swap the `k_sem_give` sequence.
closed 2023-07-12 by carlescufi