zephyrproject-rtos/zephyr #60148: boards: Add support for XIAO ESP32S3 Sense

kartben, 2023-07-07:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Seeed Studio XIAO ESP32S3 was recently added (PR #59732). As this board also has a "Sense" variant adding an OV2640 camera sensor, a microphone, and an SD card reader, it would be nice to have it supported as well.

**Describe why you are asking for this support?**

Having an affordable board with built-in camera supported upstream can only be a good thing, right? Plus, this is a good opportunity for someone to make their first contribution to the project as a lot of the heavy lifting was already done for the XIAO ESP32S3.

**Additional context**

* https://www.seeedstudio.com/XIAO-ESP32S3-Sense-p-5639.html
* [Sense Schematic](https://files.seeedstudio.com/wiki/SeeedStudio-XIAO-ESP32S3/res/XIAO_ESP32S3_ExpBoard_v1.0_SCH.pdf)
* [Pinout](https://files.seeedstudio.com/wiki/SeeedStudio-XIAO-ESP32S3/res/XIAO_ESP32S3_Sense_Pinout.xlsx)

/cc @Maxwelltoo

Maxwelltoo, 2023-07-10:
This is also what I want to do.  After trying, the microphone and SD card  works, but I believe the camera driver in Zephyr still needs an update from Espressif (the driver in esp-idf is good, but it's based on freeRTOS). So just wait...

rftafas, 2023-07-14:
This sensor has 2 interfaces: control and image. IIRC, the driver on Zephyr is the control interface. Camera interface is not supported for any ESP32 device yet. It is correct, this needs to be implemented.

Even though people can reuse/port the code from https://github.com/espressif/esp32-camera for the lower parts of the driver, I don't consider this good for 'first issue'.

pblxptr, 2024-09-14:
Any update on this issue? From what I've seen the support for LCD_CAM driver has already been merged.

lit-af, 2025-09-04:
Sorry for hi-jacking this thread. This board from Freenove would also be a good candidate for an in-tree camera board. https://store.freenove.com/products/fnk0085.

Is there anything missing in the hal_espressif at this moment that could prevent to integrate them?

closed 2025-09-04 by kartben